lardemua/atom #767: Error when calibrating with a 3d lidar.

manuelgitgomes, 2023-12-04:
Hello @miguelriemoliveira.

I am having an error when trying to calibrate softbot2:

```bash
Traceback (most recent call last):
  File "/home/nel/distrobox/ubuntu20/core_ws/devel/lib/atom_calibration/calibrate", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/nel/distrobox/ubuntu20/core_ws/src/atom/atom_calibration/scripts/calibrate", line 588, in <module>
    main()
  File "/home/nel/distrobox/ubuntu20/core_ws/src/atom/atom_calibration/scripts/calibrate", line 528, in main
    opt.callObjectiveFunction()
  File "/home/nel/distrobox/ubuntu20/core_ws/src/atom/atom_core/src/atom_core/optimization_utils.py", line 290, in callObjectiveFunction
    return self.internalObjectiveFunction(self.x)
  File "/home/nel/distrobox/ubuntu20/core_ws/src/atom/atom_core/src/atom_core/optimization_utils.py", line 311, in internalObjectiveFunction
    errors = self.errorDictToList(self.objective_function(self.data_models))
  File "/home/nel/distrobox/ubuntu20/core_ws/src/atom/atom_core/src/atom_core/optimization_utils.py", line 359, in errorDictToList
    raise ValueError('Objective function returned dictionary with residual ' + Fore.RED +
ValueError: Objective function returned dictionary with residual c000_p_pattern_1_lidar3d_oe_29 which does not exist. Use printResiduals to check the configured residuals

```

I made some changes in the code to reach this points, which I will push them to a branch named TBD.

This was done with this dataset:
[simple_train.zip](https://github.com/lardemua/atom/files/13548798/simple_train.zip)

miguelriemoliveira, 2023-12-04:
Hi @manuelgitgomes ,

I will pick it up as soon as possible, but I am not sure if that's going to be tomorrow.

miguelriemoliveira, 2023-12-05:
Hi @manuelgitgomes ,

I created the rlbot, a system with an rgb camera and a 3d lidar.  I ran into several problems due to the change of using multiple calibration patterns, but I was able to fix them all.

The calibration of lidar3d is back online!

closed 2023-12-05 by miguelriemoliveira