lardemua/atom #781: Create an atom example rgb rgb with multiple patterns (rrmpbot)

miguelriemoliveira, 2023-12-12:


miguelriemoliveira, 2023-12-14:
Its better to have an example with a camera on a manipulator. Changing the name to rihmpbot ...

miguelriemoliveira, 2023-12-23:
Drawing all patterns in the calibrate script

[image]

miguelriemoliveira, 2023-12-23:
Also able to draw images with projections for when only a single pattern is detected

[image]

or in collections where all patterns are detected

[image]

miguelriemoliveira, 2023-12-23:
Calibration using multiple patterns works fine:

Here's the system

[image]

and calibration results:

[image]

closed 2023-12-23 by miguelriemoliveira

manuelgitgomes, 2023-12-28:
These are great news! Happy to see atom advancing with such strength

miguelriemoliveira, 2023-12-28:
Now we have more time to work on it ...

commit a2af1a2f (2023-12-23): Close #781 . Calibration upd and running, readme udated.