lardemua/atom #789: Split data_collector into several ros nodes

miguelriemoliveira, 2023-12-16:
In #788, we had the idea before of splitting the collector node into several ros nodes, instead of using different threads.
We also thought about this some years ago. The reason was because we had some traffic problems in the ros message (#36), which were solved not very well. 

The idea we had was to put make each labeler  a different ros node which publishes to the new, much more simple data collector.

To synchronize the labels with the raw data, I think we should have label messages that contain both the raw data and the labels.

This will simplify the code before moving on to adding multiple pattern support #788.

miguelriemoliveira, 2023-12-17:
RGB labeling node complete.

miguelriemoliveira, 2023-12-17:
[image]

miguelriemoliveira, 2023-12-17:
Works for rrcbot 

[image]

miguelriemoliveira, 2023-12-17:
Fails for rrbot for some reason

Despite have good marker detections

[image]

The charuco corners are not well interpolated

[image]

miguelriemoliveira, 2023-12-17:
Ahh, for this pattern in particular the m*_*#$"ther fu***#"$"er needs the board to be defined like this:

```python
# self.board = cv2.aruco.CharucoBoard_create(size["y"] + 1, size["x"] + 1, length, marker_length,
            #    self.dictionary)
self.board = cv2.aruco.CharucoBoard_create(size["x"] + 1, size["y"] + 1, length, marker_length,
                                                       self.dictionary)
```

and with it a perfect detection:

[image]

miguelriemoliveira, 2023-12-17:
Ok, got it. It was a problem with the config files. Now it works fine for all patterns

[image]

miguelriemoliveira, 2023-12-18:
lidar3d labeler on the way ... 

[image]

miguelriemoliveira, 2023-12-18:
Lidar3d labeler already publishes a labels msg.

miguelriemoliveira, 2023-12-20:
Depth labeling working fine as an standalone node. Will now work on expanding the atom msgs to include depth labels.

miguelriemoliveira, 2023-12-20:
Now publishing the labels as ros messages.

miguelriemoliveira, 2023-12-20:
Now I need to tweak the data collector to subscribe these labels messages ...

miguelriemoliveira, 2023-12-20:
Actually, first I will work on making the configure calibration package script create a launch file that calls the well configured labelers ...

miguelriemoliveira, 2023-12-20:
Done. We are launching the labelers automatically upon calibration package configuration.

miguelriemoliveira, 2023-12-20:
Now we need to change the data collector to:

1. subscribe the labels topic published by the labelers, instead of the raw data streaming from the sensors
2. Receive the labels messages and convert those to an atom dataset format.

miguelriemoliveira, 2023-12-22:
I think this is complete. 

I changed a lot in the **data_collector_and_labeler.py**, now **collect data_collector.py** as it does not label anything anymore.

Subscribed to the label messages produced by the labelers.

Converted the atom_msgs containing the labels to the atom dataset format.

Tested producing a dataset and then calibrating with that dataset in rrbot, rlbot and rdbot.

All seems to be working fine. 

Furthermore, the compatibility of datasets is maintained, so we do not have to delete previous datasets.

miguelriemoliveira, 2023-12-22:
@manuelgitgomes , since I did extensive tests with the atom examples I am confident this should be working and will merge the PR right away. If some problem appears I will be on it in no time.

To use the new version a reconfiguration of the calibration packages is necessary.

closed 2023-12-22 by miguelriemoliveira

manuelgitgomes, 2023-12-28:
Hello @miguelriemoliveira!

Great job, I will be testing this soon enough so I can create new issues with bugs. :-)

miguelriemoliveira, 2023-12-28:
> Great job, I will be testing this soon enough so I can create new issues with bugs. :-)

Right. To be expected ...