lardemua/atom #800: Bug in ground truth evaluation

miguelriemoliveira, 2023-12-27:
Hi @manuelgitgomes ,

when using the **ground_truth_frame_evaluation** script, I was trying to run this:

    rosrun atom_evaluation ground_truth_frame_evaluation -train_json atom_calibration.json -test_json dataset.json -sf tripod_center_support -tf rgb_world_link

The links I am using exist, here's a picture of the tf tree

[image]

However, I get this error when running the above command:

    Source frame tripod_center_support in not a known frame. Use names matching the ones on the config.yaml calibration file

The question is why do we need to use transforms from the config yml? We could use any transform, right?

miguelriemoliveira, 2023-12-28:
Hi @manuelgitgomes ,

I was looking at your script and there are several things I do not understand. When you can, lets have a zoom talk to discuss. I had a different  idea of what you were doing, and now I am not sure if another evaluation script should be created or not ...

manuelgitgomes, 2023-12-28:
Hello @miguelriemoliveira.

This script was fully rewritten by @JorgeFernandes-Git, so I also do not have the full idea on how it is implemented.

We can possibly meet this afternoon, I will give you more details.

miguelriemoliveira, 2023-12-28:
Hi @manuelgitgomes ,

thanks. Email me when you're available.

In the meantime I will start writing a new script to do what I needed.

miguelriemoliveira, 2023-12-28:
Added a new script called  **ground_truth_transform_evaluation**

commit 0a643496 (2023-12-28): Close #800
closed 2023-12-28 by miguelriemoliveira