lardemua/atom #803: Should we have position_bias and origin_yaw parameters

miguelriemoliveira, 2023-12-29:
@manuelgitgomes FYI

They seem to do the same, so we could remove the position_bias.

I did an experiment that is to run a bagfile and look at the coordinate frame of the forearm link

The thick. short ref frame in the image.

Them I used this code to subscribe to joint state messages and produce the corresponding transformations according to the joint angles.  I did this with a change in the angle in two methods:

1. Added 45 degrees more to the position_bias parameter (the medium thickness, medium length coordinate frame in the image)
2.  Added 45 degrees to the value of the origin_yaw (the thin thickness, longer frame in the image)

[image]

As expected, both methods 1 and 2 are 45 degrees deviated from the original frame. Also, as expected, both methods result in the coincident  coordinate frames ... So it seems indeed both parameters will do the same. 

The question is why did the optimizer does not work as well when using the origin_yaw instead of the position_bias ...

miguelriemoliveira, 2023-12-29:
Got it. In the shoulder method 1 gives what was expected, but method 2 does not.

The problem occurs in the shoulder, not the elbow. I think it has to do with the values of origin_xyz and origin roll pitch different from 0. 

[image]

[image]

Will investigate later.

miguelriemoliveira, 2023-12-29:
Found the problem. We were using euler angle axis **sxyz**  and it should be **rxyz**.

Now everything makes sense:

[image]

miguelriemoliveira, 2023-12-29:
Now we go back to the experiment of trying to calibrate using position bias vs using origin_yaw ...

miguelriemoliveira, 2023-12-29:
So the calibration experiments turned out to show what we already expected: Using position bias or using origin_z (for revolute joints in which the axis is 0 0 1) gives exactly the same result.

So I think that we should remove the position_bias parameter from the optimization, and instead use the origin_yaw.

@manuelgitgomes you agree? Should I proceed with this change?

manuelgitgomes, 2023-12-30:
Hello @miguelriemoliveira!

Great news.
I think we can proceed then!

miguelriemoliveira, 2024-01-02:
Thanks @manuelgitgomes . I will proceed.

closed 2024-01-02 by miguelriemoliveira