lardemua/atom #809: Remove joint_position_bias parameter from calibration

miguelriemoliveira, 2024-01-06:
After the analysis in #803 we came to the conclusion that the joint_position_bias parameter is redundant with one other joint parameter, depending on the type and axis configuration of the joint.

For example, for a revolute joint that rotates around the z axis (i.e. axis = 0 0 1), the origin_yaw parameter does exactly the same as the joint_position_parameter. The same occurs for prismatic joints. 

The decision is to remove the joint_position_bias parameter from the optimization.

After this the atom example 

https://github.com/lardemua/atom/tree/noetic-devel/atom_examples/jcpbbot

should be revised.

miguelriemoliveira, 2024-01-06:
How to change the production of xacro at

https://github.com/lardemua/atom/blob/4022f412d06c16235f5903c22d9e4a0094458398/atom_core/src/atom_core/xacro_io.py#L70-L88

Not really sure.

miguelriemoliveira, 2024-01-06:
@manuelgitgomes I think you will have to rerun your joint based calibrations. Sorry!

closed 2024-01-06 by miguelriemoliveira

manuelgitgomes, 2024-01-08:
Great job @miguelriemoliveira!

The latest calibration on #706 was done without position bias already!