lardemua/atom #814: calibration configuration stuck

Kazadhum, 2024-01-06:
Hello! I'm trying to configure the calibration of the `zau2` system ([repo here](https://github.com/Kazadhum/zau2)) this time with a real dataset. However, when running:

`rosrun zau2_real_calib configure -utf`

the process gets stuck after displaying the NX graph, and eventually times out.

I haven't done this since the latest update to the `config` files for ATOM's calibration, so I don't know if I'm doing anything wrong (looking at the `atom_examples`, it doesn't seem that way). For reference, the configuration file is in the `zau2_real_calib` package.

Tagging @miguelriemoliveira and @manuelgitgomes for visibility.

miguelriemoliveira, 2024-01-06:
Hi @Kazadhum ,

sounds like a line I have in configure calibration package to show the graph which I use for debug. 
But it should be commented ... do you have the latest atom verstion?

manuelgitgomes, 2024-01-08:
Hello @Kazadhum 

In a quick look, your config seems ok!

Just a tip, be more descriptive in your commit messages!

Kazadhum, 2024-01-10:
Thank you for the replies @miguelriemoliveira and @manuelgitgomes!

> Hi @Kazadhum ,
> 
> sounds like a line I have in configure calibration package to show the graph which I use for debug. But it should be commented ... do you have the latest atom version?

Just checked. I do have the latest version, but I think I have found the issue... It seems like there is possibly a difference between the URDFs of the real Zau used by @JorgeFernandes-Git and the one I have built for simulation. Since the link names are different the tree isn't connected and ATOM gets stuck at the display (until I close the display, at which point ATOM raises an error saying just that). I'll continue to investigate this further and try to fix it.

> Hello @Kazadhum
> 
> In a quick look, your config seems ok!
> 
> Just a tip, be more descriptive in your commit messages!

Thank you! I'll try to improve on this! 😅 

-------------

Since this bug turned out to not be real and instead is a problem with the robot's description, I think this can be closed for now and open a new issue to address the actual problem if necessary.

closed 2024-01-10 by miguelriemoliveira