lardemua/atom #821: Improve calibration of riwmpbot using multiple patterns

miguelriemoliveira, 2024-01-09:
So far, results are:

using all three calibration patterns:

```
Parameters:
                                            V ini   V calib      V GT     E ini    E fin
tripod_center_support-rgb_world_link_x   0.000000  0.000059  0.000000  0.000000  0.00006
tripod_center_support-rgb_world_link_y   0.000000  0.000230  0.000000  0.000000  0.00023
tripod_center_support-rgb_world_link_z   0.030000  0.029918  0.030000  0.000000  0.00008
tripod_center_support-rgb_world_link_r1  0.000000  0.000031  0.000000  0.000000  0.00003
tripod_center_support-rgb_world_link_r2  0.000000  0.000495  0.000000  0.000000  0.00050
tripod_center_support-rgb_world_link_r3  0.000000 -0.000668  0.000000  0.000000  0.00067
forearm_link-charuco_170x100_3x6_x      -0.199164 -0.200086 -0.200000  0.000836  0.00009
forearm_link-charuco_170x100_3x6_y       0.020514  0.019920  0.020000  0.000514  0.00008
forearm_link-charuco_170x100_3x6_z      -0.031530 -0.029855 -0.030000  0.001530  0.00015
forearm_link-charuco_170x100_3x6_r1     -0.006031  0.004199  0.000000  0.006031  0.00420
forearm_link-charuco_170x100_3x6_r2      0.020879 -0.000571  0.000000  0.020879  0.00057
forearm_link-charuco_170x100_3x6_r3      3.139064  3.140317  3.140000  0.000936  0.00032
flange-charuco_200x200_8x8_x             0.059193  0.060027  0.060000  0.000807  0.00003
flange-charuco_200x200_8x8_y            -0.059013 -0.059782 -0.060000  0.000987  0.00022
flange-charuco_200x200_8x8_z            -0.000259  0.000272  0.000000  0.000259  0.00027
flange-charuco_200x200_8x8_r1           -0.002888  0.007277  0.000000  0.002888  0.00728
flange-charuco_200x200_8x8_r2            0.006682  0.001488  0.000000  0.006682  0.00149
flange-charuco_200x200_8x8_r3           -0.000239 -0.000280  0.000000  0.000239  0.00028
upper_arm_link-charuco_200x120_3x6_x    -0.240449 -0.240103 -0.240000  0.000449  0.00010
upper_arm_link-charuco_200x120_3x6_y     0.030525  0.029949  0.030000  0.000525  0.00005
upper_arm_link-charuco_200x120_3x6_z     0.090573  0.090234  0.090000  0.000573  0.00023
upper_arm_link-charuco_200x120_3x6_r1    0.000603  0.000410  0.000000  0.000603  0.00041
upper_arm_link-charuco_200x120_3x6_r2    0.003121  0.001123  0.000000  0.003121  0.00112
upper_arm_link-charuco_200x120_3x6_r3    3.140237  3.140346  3.140000  0.000237  0.00035
joint-elbow_joint-origin_yaw            -0.030000  0.000280  0.000000  0.030000  0.00028
joint-shoulder_lift_joint-origin_yaw     0.034000 -0.000232  0.000000  0.034000  0.00023
joint-shoulder_pan_joint-origin_yaw     -0.010000 -0.000401  0.000000  0.010000  0.00040
joint-wrist_1_joint-origin_yaw           0.050000 -0.000091  0.000000  0.050000  0.00009
joint-wrist_2_joint-origin_yaw           0.010000  0.000038  0.000000  0.010000  0.00004
joint-wrist_3_joint-origin_yaw           3.111593  3.131870  3.141593  0.030000  0.00972
```

And using only the hand calibration pattern:

```
Parameters:
                                            V ini   V calib      V GT     E ini    E fin
tripod_center_support-rgb_world_link_x   0.000000  0.000147  0.000000  0.000000  0.00015
tripod_center_support-rgb_world_link_y   0.000000  0.000096  0.000000  0.000000  0.00010
tripod_center_support-rgb_world_link_z   0.030000  0.029889  0.030000  0.000000  0.00011
tripod_center_support-rgb_world_link_r1  0.000000  0.000039  0.000000  0.000000  0.00004
tripod_center_support-rgb_world_link_r2  0.000000  0.000471  0.000000  0.000000  0.00047
tripod_center_support-rgb_world_link_r3  0.000000 -0.000539  0.000000  0.000000  0.00054
flange-charuco_200x200_8x8_x             0.059193  0.060021  0.060000  0.000807  0.00002
flange-charuco_200x200_8x8_y            -0.059013 -0.059793 -0.060000  0.000987  0.00021
flange-charuco_200x200_8x8_z            -0.000259  0.000255  0.000000  0.000259  0.00025
flange-charuco_200x200_8x8_r1           -0.002888  0.007524  0.000000  0.002888  0.00752
flange-charuco_200x200_8x8_r2            0.006682  0.001403  0.000000  0.006682  0.00140
flange-charuco_200x200_8x8_r3           -0.000239 -0.000306  0.000000  0.000239  0.00031
joint-elbow_joint-origin_yaw            -0.030000  0.000302  0.000000  0.030000  0.00030
joint-shoulder_lift_joint-origin_yaw     0.034000 -0.000188  0.000000  0.034000  0.00019
joint-shoulder_pan_joint-origin_yaw     -0.010000 -0.000277  0.000000  0.010000  0.00028
joint-wrist_1_joint-origin_yaw           0.050000 -0.000080  0.000000  0.050000  0.00008
joint-wrist_2_joint-origin_yaw           0.010000  0.000043  0.000000  0.010000  0.00004
joint-wrist_3_joint-origin_yaw           3.111593  3.131620  3.141593  0.030000  0.00997
```

The improvement of using 3 patterns is not so very big ...

miguelriemoliveira, 2024-01-12:
Good news. I think now we are getting the results we expected: Better results when using multiple patterns.

**Using only the hand pattern**

[image]

**Using the three patterns**

[image]

These are good news. Before I am not sure why it was not working. I could just be that the results were very hard to inspect and that I misread them.

manuelgitgomes, 2024-01-12:
Hello @miguelriemoliveira, why is the error on shoulder pan so high?

miguelriemoliveira, 2024-01-12:
> why is the error on shoulder pan so high?

I suspect it is because that joint does not move a lot in the dataset.  In order for the camera to view the patterns, the arm must be in a special configuration where the shoulder pan does not move a lot. My explanation is that is the reason.

I will try to make an experiment that proves it.

miguelriemoliveira, 2024-01-12:
Here's a table with statistics about the joint positions:

[image]

I was expecting that the should pan would be the joint with least observed excursion, but that is the shoulder lift joint.

So my explanation above perhaps does not make a lot of sense ...

miguelriemoliveira, 2024-01-13:
This is the image with all the collections.

[image]

From the observation of the image the 15 degrees range for the shoulder lift look ok

[image]

For the shoulder pan 

[image]

miguelriemoliveira, 2024-01-13:
I think this issue is complete. We've shown that using multiple patterns actually improves the performance. Will try to run several experiments in batch to have statistical significance.

closed 2024-01-13 by miguelriemoliveira