lardemua/atom #826: Error when running generate_joint_space_tsne

miguelriemoliveira, 2024-01-12:
When running

    rosrun atom_calibration generate_joint_space_tsne -json ./dataset.json

I get this error

```
Traceback (most recent call last):
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_calibration/scripts/utilities/generate_joint_space_tsne", line 76, in <module>
    main()
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_calibration/scripts/utilities/generate_joint_space_tsne", line 40, in main
    raise ValueError(f'Residuals file {residuals_path} does not exist.')
ValueError: Residuals file ./calibration_results.csv does not exist.
```

manuelgitgomes, 2024-01-12:
We could add an option to just see the residuals optionally.

miguelriemoliveira, 2024-01-12:
Perhaps it's better.  Can you do it?

In any case I am creating a table on the side of the inspect_atom_dataset. So its not a priority.