lardemua/atom #835: ATOM calibration package configuration static transform check misses 'static' joint type

Kazadhum, 2024-02-11:
Hello! I've been trying to configure the calibration of the zau2 system with real data and I've encountered a problem.

Running the command:

`rosrun zau2_real_calib configure -utf`

returns the error:

`ATOM Error: Sensor hand_camera defines transformation hand_camera_link to hand_camera_rgb_frame to be calibrated but this transformation is not static and therefore cannot be calibrated.`

I've checked ATOM's source code for where it fetches the info regarding the joints, which is the URDF. I then checked the URDF to see what type is specificied for this joint and it is `fixed`. For reference, [here is the snippet of code where ATOM checks if a TF is static](https://github.com/lardemua/atom/blob/8b43c60ba7119b8d9823c9dbb6a38e02106867ea/atom_calibration/scripts/configure_calibration_pkg#L415).

I added a `print(edge['type'])` for debugging and instead of `'fixed'`, I got `static`. If this worked before, and `edge['type']` used to hold the value `'fixed'`, I don't know why it changed to `static`. It shouldn't have anything to do with the URDF, since it is similar (and this particular joint in defined in a separate file from `mmtbot` which I have not altered). It should also not be related to the bag file, since ATOM fetches this info from the URDF. So, I'm unsure as to what the cause for this might be.

As a bit of context (which I don't think will be relevant to this issue), I've had to change the names of the TF frames present in the original bag file to fit with the URDF of the simulated system I've built. I did this with the `rename_tf_frames` script I wrote for the [rosbag_tools repo](https://github.com/miguelriemoliveira/rosbag_tools).

For now, I've changed [this line of code](https://github.com/lardemua/atom/blob/8b43c60ba7119b8d9823c9dbb6a38e02106867ea/atom_calibration/scripts/configure_calibration_pkg#L420) to:

`if not edge['type'] in ['fixed', 'static']:`

as a temporary fix, hoping it's correct. I don't know if anyone else has encountered this issue and wanted to check before I push any changes to the ATOM repo. @miguelriemoliveira, @manuelgitgomes, @danifpdra any thoughts?

Here's [the link to the zau2 repo](https://github.com/Kazadhum/zau2).

Kazadhum, 2024-02-11:
I found the problem. When using the `-utf/--use_tfs` flag, ATOM builds the NX graph of nodes from the bagfile. Then it adds an edge to that graph for each node according to the topic where the TF is published: /tf or /tf_static. If it's published under /tf_static, it gives that graph edge the type 'static', instead of 'fixed'.

So, in [config_visualization.py](https://github.com/lardemua/atom/blob/8b43c60ba7119b8d9823c9dbb6a38e02106867ea/atom_core/src/atom_core/config_visualization.py#L143-L157) I can change:

```
for topic, msg, t in bag.read_messages(topics=['/tf_static']):
    for transform in msg.transforms:
        if not nx_graph.has_edge(transform.header.frame_id.replace('/', ''), transform.child_frame_id.replace('/', '')):
            # print(transform.header.frame_id.replace('/',''), transform.child_frame_id.replace('/',''))
            nx_graph.add_edge(transform.header.frame_id.replace('/', ''),
                              transform.child_frame_id.replace('/', ''), weight=1, type='static')
```

to:

```
for topic, msg, t in bag.read_messages(topics=['/tf_static']):
    for transform in msg.transforms:
        if not nx_graph.has_edge(transform.header.frame_id.replace('/', ''), transform.child_frame_id.replace('/', '')):
            # print(transform.header.frame_id.replace('/',''), transform.child_frame_id.replace('/',''))
            nx_graph.add_edge(transform.header.frame_id.replace('/', ''),
                              transform.child_frame_id.replace('/', ''), weight=1, type='fixed')
```

to be in conformity with [this check](https://github.com/lardemua/atom/blob/8b43c60ba7119b8d9823c9dbb6a38e02106867ea/atom_calibration/scripts/configure_calibration_pkg#L414-L432).

miguelriemoliveira, 2024-02-12:
Hi @Kazadhum ,

the fix sounds good, but I am afraid that we will use the static or fixed keyword elsewhere. 

I remember for example here 

https://github.com/lardemua/atom/blob/8b43c60ba7119b8d9823c9dbb6a38e02106867ea/atom_core/src/atom_core/config_visualization.py#L127-L132

we use fixed.

My suggestion is: Lets use only the "fixed" and eliminate the use of the static keyword. 
Push the fix to ATOM and if something breaks we'll know.

Another suggestion is to search atom for the "static" keyword.

Kazadhum, 2024-02-12:
Okay! Since I was working on a different branch I'll just make a pull request and merge it then. Should we close this issue for now or keep it open for now @miguelriemoliveira?

Kazadhum, 2024-02-12:
Sorry for the confusion. When I merged the pull request it closed this issue automatically.

miguelriemoliveira, 2024-02-12:
Yes, better to close. Great work!

closed 2024-02-12 by Kazadhum