lardemua/atom #836: ATOM doesn't properly build the DotGraph in config_visualization.py

Kazadhum, 2024-02-12:
I'm still doing some debugging and I will describe the issue in greater detail soon.

Kazadhum, 2024-02-12:
I understand the source of the issue now.

The `createNxGraph()` function, when the `-utf/--use_tfs` flag isn't used, does the following:

https://github.com/lardemua/atom/blob/07c011a95af53bbf37dc6231df1fc33aca6fdb8e/atom_core/src/atom_core/config_visualization.py#L104-L108

As we can see, it creates a node for each link in the robot's description and gives that node a set of properties. These properties are then used by the `createDotGraph()` function.

However, when using the `-utf` flag, `createNxGraph()` instead generates the edges without first generating the nodes:

https://github.com/lardemua/atom/blob/07c011a95af53bbf37dc6231df1fc33aca6fdb8e/atom_core/src/atom_core/config_visualization.py#L143-L157

Since the nodes aren't attributed that set of properties, the `createDotGraph()` function runs into a KeyError. This happened to me while running the configuration for the calibration of the Zau2 system with real data, where I had to use the `-utf` flag.

Running `rosrun zau2_real_calib configure -utf` resulted in:
```
Traceback (most recent call last):
  File "/home/diogo/catkin_ws/devel/lib/atom_calibration/configure_calibration_pkg", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/diogo/catkin_ws/src/atom/atom_calibration/scripts/configure_calibration_pkg", line 528, in <module>
    dot_graph = createDotGraph(nx_graph, config)
  File "/home/diogo/catkin_ws/src/atom/atom_core/src/atom_core/config_visualization.py", line 183, in createDotGraph
    if node['is_world']:
KeyError: 'is_world'

exit status: 1
```

I'll try to correct this as soon as I can.

Kazadhum, 2024-02-12:
I changed the `createNxGraph()` function to give the attributes to the nodes after creating the graph from the TFs in the bagfile:

https://github.com/lardemua/atom/blob/7eab33a776e589035017f52674f2f49ac8e32ae7/atom_core/src/atom_core/config_visualization.py#L143-L171

However, I ran into another issue now:

```
Traceback (most recent call last):
  File "/home/diogo/catkin_ws/devel/lib/atom_calibration/configure_calibration_pkg", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/diogo/catkin_ws/src/atom/atom_calibration/scripts/configure_calibration_pkg", line 528, in <module>
    dot_graph = createDotGraph(nx_graph, config)
  File "/home/diogo/catkin_ws/src/atom/atom_core/src/atom_core/config_visualization.py", line 254, in createDotGraph
    dot_graph.edge(edge['parent'], edge['child'], color=rgb, style='solid',
KeyError: 'parent'

exit status: 1
```

Looking into it now.

Kazadhum, 2024-02-12:
It seems as though the same issue happens with the edges as it did with the nodes, where the edges' attributes are not defined. So, even though the edge is defined using the parent and child link's name, the edge is not given either the "parent" or "child" attribute. For comparison:

Here is the snippet where each graph edge is created when `-utf` is *not* used:
https://github.com/lardemua/atom/blob/ea0c30e72a28e1ec1959170ccf8a6145929aa46b/atom_core/src/atom_core/config_visualization.py#L134-L141

And here is where they are created when `-utf` is used:
https://github.com/lardemua/atom/blob/ea0c30e72a28e1ec1959170ccf8a6145929aa46b/atom_core/src/atom_core/config_visualization.py#L152-L157

Kazadhum, 2024-02-12:
I've made some changes. I've changed the creation of the edges when using `-utf` to the following:

https://github.com/lardemua/atom/blob/e6d7b7e44b6d130a37657483575de64f8250fc05/atom_core/src/atom_core/config_visualization.py#L146-L164

Note that some attributes are still missing when compared to the non-utf case:  `joint_params_calibrated` and `joint_name`. I am unsure as to how to include them, since the joints come from the URDF file and not the bagfile. 

Regardless, running `rosrun zau2_real_calib configure -utf` now does not result in an error. The configuration runs successfully. However, here is the diagram that results from this:

[image]

As you can see, all of the non-fixed or static joints are regarded as "unknown cases". I've spotted the cause for this back to the creation of the edges.

miguelriemoliveira, 2024-02-12:
Hi @Kazadhum , great work. I will try to respond.

> However, I don't think this kind of information is present in the bagfile. How should I proceed? Additionally, does this information have any bearing in the actual calibration?

Well, we use it for inspection of the pdf, but we also use during calibration. For example, if we have a non-fixed pattern we get multiple transformations.

https://github.com/lardemua/atom/blob/ea0c30e72a28e1ec1959170ccf8a6145929aa46b/atom_calibration/scripts/calibrate#L371-L376
 
> Another question I have that has me questioning this whole thread is the following: if this was an issue with ATOM, why haven't I had this same issue when I used the -utf flag for some other calibrations before? 
I also assume that other people have used -utf without problems. So, why were these changes necessary now?

That should be my fault :-( I change the createdotgraph and similar functions only some weeks ago. 

This of ATOM as a living being, constantly changing and evolving.  A very messy one I should add : -)

miguelriemoliveira, 2024-02-12:
For now in my opinion just move on. I have an idea of merging the alternatives -utf and don't  use tr into a single approach. First we look in the urdf, then if we can't find the information there we look into the bagfile.

I think this would solve these issues.

Created #838 for this.

miguelriemoliveira, 2024-02-12:
>  I couldn't do it while doing my master's but maybe now I can dedicate some time to begin writing them. Couldn't hurt.

Definitely. And we have atom examples which should be useful in this regard.

closed 2024-02-12 by Kazadhum