lardemua/atom #842: riwbot examples data fails to download

brunofavs, 2024-02-14:
There seems to be a issue downloading riwbot example data when executing:

`rosrun atom_calibration download_atom_examples_data`

[image]

All other examples downloaded with any issue.

I reckon it has something to do the drive yet I can't be sure.

manuelgitgomes, 2024-02-14:
Hello @brunofavs.
Can you show your downloaded atom_examples?

The error comes from wanting to unzip a zipped dataset but not being able to.

manuelgitgomes, 2024-02-14:
Strange, might be something corrupted in that specific file?

But I have downloaded it as you did and nothing was wrong.
Here it is, see if this works.
[train.zip](https://github.com/lardemua/atom/files/14279510/train.zip)

manuelgitgomes, 2024-02-14:
I have this downloaded a long time ago, I will test this further on a newly downloaded copy.

miguelriemoliveira, 2024-02-14:
Thanks @brunofavs for posting the issue.

I had the wrong link, not sure why. Can you test again please?

brunofavs, 2024-02-14:
This time everything worked as expected, without any errors.

I'm also having some trouble launching the calibration studio because the charuco model isn't loading in but I'll try to figure it out and if I can't will post a issue later.

Meanwhile I think this issue is solved, I'm closing it now

closed 2024-02-14 by brunofavs