lardemua/atom #843: Charuco and chessboard patterns not loading in calibration_studio

brunofavs, 2024-02-15:
Hey,

Whenever I try to launch either the calibration_studio.world or the calibration_studio_with_chessboard.world

`roslaunch rrbot_gazebo gazebo.launch` (charucos)

or 

`roslaunch rrcbot_gazebo gazebo.launch` (chessboard)

I'm getting the following errors : 
[image]
[image]

And no pattern is loaded in the world :

[image]

I wasn't familiar with how URI's worked before so I was studying a bit and found this :

https://github.com/lardemua/atom/blob/64465f7994f4e2c12cd0dd684df35a8b7d73299a/atom_core/src/atom_core/config_io.py#L155-L176

And I'm confused as there is no "model"  URI scheme defined anywhere.

I checked and I have the necessary files in my local repo

[image]

But I cannot get it to load.

However it's funny because when I bring up the robot both the robot and pattern show on rviz however just the model shows on gazebo.

[image]

This truly feels like a newbie issue tbh.

manuelgitgomes, 2024-02-15:
Hello @brunofavs!

Yes this is a newbie issue :)
But to be fair to you, I quickly searched through the documentation and did not find anything in there on how to solve this.

Gazebo has a property called `GAZEBO_MODEL_PATH` in which you define the models that gazebo can load.
Since you did not update it to have `atom_worlds/pattern/models`, Gazebo searches for it and does not find it.

To add it, add this line to your .bashrc or .zshrc:
```bash
export GAZEBO_MODEL_PATH="`rospack find atom_worlds`/pattern/models:${GAZEBO_MODEL_PATH}"
```
This will update the model path to have the models on atom_worlds.

Another thing, very nice Issue construction, with a lot of details.
Just one caveat, on your error messages, please copy and paste them instead of adding a print screen.
Having it in text makes it easy for other people to search your issue on the web.

miguelriemoliveira, 2024-02-15:
This information used to be in the installation section of each robotic system, but its not in atom examples.

@brunofavs , please create an issue to add it to the documentation?

> Another thing, very nice Issue construction, with a lot of details.
> Just one caveat, on your error messages, please copy and paste them instead of adding a print screen.
> Having it in text makes it easy for other people to search your issue on the web.

and also easier to read. Not everyone likes a dark background, and has your preference for font size ...

brunofavs, 2024-02-15:
Hey @miguelriemoliveira @manuelgitgomes 
Thank you both for the help, I added 

> To add it, add this line to your .bashrc or .zshrc:
>export GAZEBO_MODEL_PATH="`rospack find atom_worlds`/pattern/models:${GAZEBO_MODEL_PATH}"

And it worked right away :)

I do remember now having these issues back in @automec too with the traffic signals.

Thanks also for the feedback regarding the issue construction, I see now why pasting prints of error messages is undesirable.

I will do a follow-up issue soon on the documentation.

closed 2024-02-15 by brunofavs