lardemua/atom #851: Error launching rihbot

brunofavs, 2024-02-27:
Hey,

I'm getting the following error trying to bringup rihbot :

`roslaunch rihbot_bringup bringup.launch
`

```
Resource not found: ur10e_moveit_config
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/home/brunofavs/atom_ws/src
ROS path [2]=/home/brunofavs/catkin_ws/src
ROS path [3]=/opt/ros/noetic/share
The traceback for the exception was written to the log fil
```

I already installed the ur10e_config package as described in the wiki 
[http://wiki.ros.org/ur10_moveit_config](url)

I think I'm missing something on my shell configuration (.zshrc).

I searched in the wiki and in the repository and didn't find anything regarding this problem.

This is everything I have in my shell configuration regarding ROS and ATOM

```
132 #---------------------- 
133 #      ROS 
134 #--------------------- 
135
136 source /opt/ros/noetic/setup.zsh 
137  
138 #---------------------- 
139 #    ROS WORKSPACES 
140 #--------------------- 
141  
142 # Only one workspace can be active at a time 
143  
144 # Mess arround workspace 
145 #source ~/catkin_ws/devel/setup.zsh 
146 # Atom workspace 
147 source $HOME/atom_ws/devel/setup.zsh 
148  
149 #---------------------- 
150 #    ATOM 
151 #--------------------- 
152  
153 alias atom="cd $HOME/atom_ws/src/atom" 
154  
155 export ROS_BAGS="$HOME/bagfiles" 
156 export ATOM_DATASETS="$HOME/datasets" 
157  
158 export GAZEBO_MODEL_PATH="`rospack find atom_worlds`/pattern/models:${GAZEBO_MODEL_PATH}" 
```

Kazadhum, 2024-02-27:
Hi @brunofavs! I think I know what the problem might be... The `ur10_moveit_config` package is probably the same one that is mentioned in the README for mmtbot, which is present in the [larcc fork of the universal_robot repo](https://github.com/lardemua/universal_robot/tree/calibration_devel). Make sure you use the `calibration_devel` branch of this repo.

So I think you just need to unninstall the ur10_moveit_config package you have and just clone the repo I've linked to your ROS workspace.

I've launched the system with no issues, so I'm pretty sure that must be it.

brunofavs, 2024-02-27:
That seemed to be the root of the problem.

It's now working I think. Not completely sure why there are 2 robots in rviz but I'll figure it out.

[image]

closed 2024-02-27 by brunofavs

manuelgitgomes, 2024-02-27:
> Not completely sure why there are 2 robots in rviz but I'll figure it out.

Hello!

Welcome to MoveIt, the open source robot manipulation tool.

Basically the "orange" robot is the end-state of the robot, that you can then call MoveIt to make the robot move to that state.
So all is good!

brunofavs, 2024-02-27:
Hey @manuelgitgomes 

Yes I've been enlightened about it by a dear colleague here in LAR :D 

Thanks!!