lardemua/atom #852: Error in getTransformationFromJoint()

brunofavs, 2024-02-27:
Hey,

When executing the calibration of rihbot, I'm getting the following error :

```
Traceback (most recent call last):
  File "/home/brunofavs/atom_ws/devel/lib/atom_calibration/calibrate", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/brunofavs/atom_ws/src/atom/atom_calibration/scripts/calibrate", line 704, in <module>
    main()
  File "/home/brunofavs/atom_ws/src/atom/atom_calibration/scripts/calibrate", line 632, in main
    residuals = objectiveFunction(opt.data_models)
  File "/home/brunofavs/atom_ws/src/atom/atom_calibration/src/atom_calibration/calibration/objective_function.py", line 399, in objectiveFunction
    quat, trans = getTransformationFromJoint(joint)
  File "/home/brunofavs/atom_ws/src/atom/atom_core/src/atom_core/joint_models.py", line 12, in getTransformationFromJoint
    if joint['joint_type'] == 'revolute' or joint['joint_type'] == 'continuous':
KeyError: 'joint_type'
```
The bug lays in this piece of code : 

https://github.com/lardemua/atom/blob/dc6c33f609ad4ec80fa3d317453cc30bd95c8626/atom_core/src/atom_core/joint_models.py#L10-L31

After printing `joint`, it does not have in fact a key `joint_type` but rather `xacro_joint_type` : 

```
{'axis_x': 0.0, 'axis_y': 0.0, 'axis_z': 1.0, 'origin_pitch': 0.0, 'origin_roll': 0.0, 'origin_x': -0.6127, 'origin_y': 0.0, 'origin_yaw': 0.0,
 'origin_z': 0.0, 'position': 0.2951597524958425, 'position_bias': 0.0, 'transform_key': 'upper_arm_link-forearm_link', 
'xacro_joint_name': 'elbow_joint', 'xacro_joint_type': 'revolute'}
```
Upon switching this keyword it seemed to work right away : 
[image]

After I checked the dataset itself and both the datasets for the riwbot and rihbot have only the `xacro_joint_type` keyword

[Uploading dataset.json…]() (here's the file as I can't get a github permalink in this case :c )

```
81       "joints": {
   82         "elbow_joint": {
   83           "axis_x": 0.0,
   84           "axis_y": 0.0,
   85           "axis_z": 1.0,
   86           "origin_pitch": 0.0,
   87           "origin_roll": 0.0,
   88           "origin_x": -0.6127,
   89           "origin_y": 0.0,
   90           "origin_yaw": 0.0,
   91           "origin_z": 0.0,
   92           "position": 0.917319427374828,
   93           "position_bias": 0.0,
   94           "transform_key": "upper_arm_link-forearm_link",
   95           "xacro_joint_name": "elbow_joint",
   96           "xacro_joint_type": "revolute"
   97         }
   98       }, 
```

As I'm not sure if this will break the pipeline elsewhere, it's on a separate branch on which I will create a pull request.

miguelriemoliveira, 2024-02-27:
Hi @brunofavs ,

thanks for the fix. Please create the PR.

brunofavs, 2024-02-28:
Fixed already, check the referenced PR for more information.

miguelriemoliveira, 2024-03-05:
Hi @brunofavs ,

me and @Kazadhum are working on riwmpbot_real and we now have a problem with this.

> raceback (most recent call last):
>   File "/home/mike/workspaces/catkin_ws/devel/lib/atom_calibration/calibrate", line 15, in <module>
>     exec(compile(fh.read(), python_script, 'exec'), context)
>   File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_calibration/scripts/calibrate", line 704, in <module>
>     main()
>   File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_calibration/scripts/calibrate", line 632, in main
>     residuals = objectiveFunction(opt.data_models)
>   File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/calibration/objective_function.py", line 399, in objectiveFunction
>     quat, trans = getTransformationFromJoint(joint)
>   File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_core/src/atom_core/joint_models.py", line 14, in getTransformationFromJoint
>     if joint['xacro_joint_type'] == 'revolute' or joint['xacro_joint_type'] == 'continuous':
> KeyError: 'xacro_joint_type'
> INFO - 2024-03-05 09:51:35,418 - core - signal_shutdown [atexit]

We checked and the correct format is "joint_type", which is being produced by the collect_data.py script

https://github.com/lardemua/atom/blob/dbb2ed00173e0e29a0eb88a5454d0c440eb39f4e/atom_calibration/src/atom_calibration/collect/data_collector.py#L466

What happened before is that you were using an old rihbot dataset which still had the old "xacro_joint_type" name.
We should create a new, corrected rihbot dataset for replacing the old one. Can you do it please?

In the meantime  we will revert your fix to use the "joint_type" field as it should.

miguelriemoliveira, 2024-03-05:
@Kazadhum just suggested that perhaps its easier to go to the rihbot dataset.json and replace the "xacro_joint_type" with "joint_type". You can try this solution and see if it works ...

closed 2024-03-05 by miguelriemoliveira

brunofavs, 2024-03-05:
Hey @miguelriemoliveira @Kazadhum 
Ok, it's settled on `joint_type` then.

I used vim's [vimgrep ](https://engineeringfordatascience.com/posts/how_to_search_and_replace_across_multiple_files_in_vim/)to substitute all instances of `xacro_joint_type` in the atom examples datasets folder to `joint_type`. It was quite a satisfactory result of brains over brawns :)

Should I send you the updated dataset folder? I don't think I can attach it here in the issue due to size.

miguelriemoliveira, 2024-03-05:
> Should I send you the updated dataset folder? I don't think I can attach it here in the issue due to size.

Yes, a zip file. I have to replace the incorrect in the drive ...

brunofavs, 2024-03-05:
When I get back to the pc I will send a wetransfer link here.

brunofavs, 2024-03-06:
Here it is @miguelriemoliveira 

I could've sent it earlier but honestly I forgot

https://we.tl/t-rqlcgjqhZL

miguelriemoliveira, 2024-03-06:
Hi @brunofavs ,

you sent. All. Which ones are to be used to replace those online?

brunofavs, 2024-03-06:
Uuh
Honestly, I have no idea right now..

Some were ok right out of the box. Some weren't, but I didn't keep track :(

brunofavs, 2024-03-06:
But in fairness, I only changed the dataset.json 's.
I sent everything because I thought it would be easier for you to just replace the folder in 1 go

brunofavs, 2024-03-06:
Im lying actually
I changed it everywhere where there was a "xacro_joint_type" keyword

This includes the atom_calibration.json for instance

miguelriemoliveira, 2024-03-06:
This is not helpful. I can't just copy all the folders and replace my folders. You have to tell me which ones ...

brunofavs, 2024-03-06:
Im not sure why that is but

I tried filtering by edit date, that didn't help me much because I redowloaded multiple times with and without overriding so it's hard to track.

Then I changed my env variables $ROS_BAGS and $ATOM_DATASETS to download again the examples to another folder to compare the files but Im getting download errors I dont know why (on rdbot)

Sorry for not having an terminal output here but Im off the pc now but wanted to give some feedback.

I'll bring it up tomorrow in the meeting

miguelriemoliveira, 2024-03-06:
perhaps using  command

    history

miguelriemoliveira, 2024-03-06:
If the rdbot does not download please create an issue

commit 6b3575a8 (2024-03-05): Fix #852