lardemua/atom #854: Error in -ctgt in mjrbot

brunofavs, 2024-02-28:
Hey everyone :)

I was trying to get this feature to work :
https://github.com/lardemua/atom/blob/dc6c33f609ad4ec80fa3d317453cc30bd95c8626/atom_examples/mrjbot/README.md#L121-L122

And it was giving errors since the flag `-pp` has been removed. I found out it was reworked and renamed in a previous commit  49e05ad to `-ctgt` by @miguelriemoliveira 

When finishing the execution with this updated flag, the code throws a error in the end : 
**Command**
```
rosrun atom_calibration calibrate -json $ATOM_DATASETS/mrjbot/train/dataset.json -ctgt \
-v -rv -nig 0.1 0.1 --phased  \
-jbn shoulder_pan_joint shoulder_lift_joint elbow_joint wrist_1_joint wrist_2_joint wrist_3_joint \
-jbp origin_yaw origin_yaw origin_yaw origin_yaw origin_yaw origin_yaw \
-jbv -0.01 0.034 -0.03 0.05 0.01 -0.03
```

**Output**
```
Traceback (most recent call last):
  File "/home/brunofavs/atom_ws/devel/lib/atom_calibration/calibrate", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/brunofavs/atom_ws/src/atom/atom_calibration/scripts/calibrate", line 704, in <module>
    main()
  File "/home/brunofavs/atom_ws/src/atom/atom_calibration/scripts/calibrate", line 671, in main
    printComparisonToGroundTruth(dataset, dataset_initial, dataset_ground_truth,
  File "/home/brunofavs/atom_ws/src/atom/atom_core/src/atom_core/utilities.py", line 211, in printComparisonToGroundTruth
    transform_ground_truth = dataset_ground_truth['collections'][selected_collection_key]['transforms'][transform_key]
KeyError: 'world-pattern_link'
```
I was sure it was again a simple keyword fix like in #852 but I found out that for some reason both the gt dataset and the initial dataset have this `world-pattern_link` (IMO it's confusing to mix both underscores and dashes in naming conventions) but for some reason the dataset original doesn't have it : 
```
datasets keys
base_link-base
base_link-base_link_inertia
base_link_inertia-shoulder_link
flange-rgb_hand_link
flange-tool0
forearm_link-wrist_1_link
rgb_hand_link-rgb_hand_optical_frame
shoulder_link-upper_arm_link
table-leg1_table
table-leg2_table
table-leg3_table
table-leg4_table
table-tabletop_table
tabletop_table-base_link
upper_arm_link-forearm_link
world-table
wrist_1_link-wrist_2_link
wrist_2_link-wrist_3_link
wrist_3_link-flange
world-pattern_link

datasets GT keys
base_link-base
base_link-base_link_inertia
base_link_inertia-shoulder_link
flange-rgb_hand_link
flange-tool0
forearm_link-wrist_1_link
rgb_hand_link-rgb_hand_optical_frame
shoulder_link-upper_arm_link
table-leg1_table
table-leg2_table
table-leg3_table
table-leg4_table
table-tabletop_table
tabletop_table-base_link
upper_arm_link-forearm_link
world-table
wrist_1_link-wrist_2_link
wrist_2_link-wrist_3_link
wrist_3_link-flange

datasets INITIAL keys
base_link-base
base_link-base_link_inertia
base_link_inertia-shoulder_link
flange-rgb_hand_link
flange-tool0
forearm_link-wrist_1_link
rgb_hand_link-rgb_hand_optical_frame
shoulder_link-upper_arm_link
table-leg1_table
table-leg2_table
table-leg3_table
table-leg4_table
table-tabletop_table
tabletop_table-base_link
upper_arm_link-forearm_link
world-table
wrist_1_link-wrist_2_link
wrist_2_link-wrist_3_link
wrist_3_link-flange
world-pattern_link
```

I dug further because I was unsure how the ground truth dataset was obtained and found it to be just a copy of the dataset before noises were added : 
https://github.com/lardemua/atom/blob/dc6c33f609ad4ec80fa3d317453cc30bd95c8626/atom_calibration/scripts/calibrate#L233-L237

But this confuses me more, if they were copies they should have the same keys, unless the ground truth was modified elsewhere and I'm not seeing it.

miguelriemoliveira, 2024-02-29:
Hi @brunofavs ,

I will try to look into this either tomorrow or friday and get back to you.

brunofavs, 2024-03-04:
Just adding that the same issue is also here :

https://github.com/lardemua/atom/blob/dbb2ed00173e0e29a0eb88a5454d0c440eb39f4e/atom_examples/spjbot/README.md#L116

miguelriemoliveira, 2024-03-05:
Hi @brunofavs ,

> But this confuses me more, if they were copies they should have the same keys, unless the ground truth was modified elsewhere and I'm not seeing it.

this only works for simulation. In simulation a dataset will contain the perfect, ground truth sensor poses.
Therefore, we can copy the dataset to a ground_truth_dataset before we add noise. The noise will modify the dataset to be used during calibration, but the ground truth dataset will remain with the original, correct values.

miguelriemoliveira, 2024-03-05:
OK, I got the problem. It is a bit tricky.

When we are defining a robotic system, we may or may not define the position of the pattern(s) in the xacro.

For example, the xacro for the mrjbot does not have the transformation from the world to the pattern defined.

https://github.com/lardemua/atom/blob/noetic-devel/atom_examples/mrjbot/mrjbot_description/urdf/robot.urdf.xacro

whereas the riwmpbot does have these transformations defined:

https://github.com/lardemua/atom/blob/0e7c3f9ed2616f640ff9ad05a11cec0e8a00be05/atom_examples/riwmpbot/riwmpbot_description/urdf/robot.urdf.xacro#L196-L200

That means that, with respect to the pose of the patterns, the mrjbot does not have ground truth and the riwmpbot does.

When using the -ctgt flag the system will compare against the ground truth data, which was did not exist in the case of mrjbot.

The solution I think is to avoid the crash, and to throw out a warning saying that it is not possible to compute the error w.r.t. the ground truth, as there isn't one.

miguelriemoliveira, 2024-03-05:
Added a crash protection that throws a warning, seen here on the first line

[image]

The other comparisons are viable, so the system continues to print those.

miguelriemoliveira, 2024-03-05:
@brunofavs please confirm its solved and close if so ...

brunofavs, 2024-03-05:
Hi @miguelriemoliveira,

> When we are defining a robotic system, we may or may not define the position of the pattern(s) in the xacro.
> 
> For example, the xacro for the mrjbot does not have the transformation from the world to the pattern defined.

Ah I see, this is why whenever I launch a world, sometimes it shows a pattern whereas other times it does not. Meaning it can defined in the xacro or the world description.

I also confirmed that this answers one of my previous "mini confusions": the pattern would only show in the live TF tree sometimes.

This raises another doubt for me... What is the difference between the 2 approaches? When should one define the pattern in the xacro or the world?

brunofavs, 2024-03-05:
> this only works for simulation. In simulation a dataset will contain the perfect, ground truth sensor poses. Therefore, we can copy the dataset to a ground_truth_dataset before we add noise. The [...]

I see, makes sense.

> @brunofavs please confirm its solved and close if so ...
I will test this now and get back to you soon.

brunofavs, 2024-03-05:
Hey
Everything seems to be working fine :) 

```
....

ATOM Warn: Cannot print comparison to ground truth for pattern pattern_1 because there is no ground truth data.

Transforms Calibration
Et: average translation error (Et0 - initial) [m],  Erot: average rotation error (Erot0 - initial) [rad]
Translation errors: < 1 mm | < 1 cm | < 5 cm | >= 5 cm
Rotation errors: < 0.1 deg | < 0.5 deg | < 1 deg | < 3 deg | >= 3 deg
+----------------------+-------------+---------+---------+-------------+------------+
|      Transform       | Description | Et0 [m] |  Et [m] | Rrot0 [rad] | Erot [rad] |
+----------------------+-------------+---------+---------+-------------+------------+
| flange-rgb_hand_link |   rgb_hand  | 0.02718 | 0.00078 |   0.09639   |  0.01203   |
|       Averages       |     ---     | 0.02718 | 0.00078 |   0.09639   |  0.01203   |
+----------------------+-------------+---------+---------+-------------+------------+
Saving transforms ground truth comparison to /home/brunofavs/datasets/mrjbot/train/results/comparison_to_ground_truth_transforms.csv

Joints Calibration
Translation errors: < 1 mm | < 1 cm | < 5 cm | >= 5 cm
Rotation errors: < 0.1 deg | < 0.5 deg | < 1 deg | < 3 deg | >= 3 deg
+---------------------+------------+-------------+---------------+
|        Joint        |   Param    | Error (ini) | Error (calib) |
+---------------------+------------+-------------+---------------+
|     elbow_joint     | origin_yaw |   0.03000   |    0.00034    |
| shoulder_lift_joint | origin_yaw |   0.03400   |    0.00049    |
|  shoulder_pan_joint | origin_yaw |   0.01000   |    0.00403    |
|    wrist_1_joint    | origin_yaw |   0.05000   |    0.00035    |
|    wrist_2_joint    | origin_yaw |   0.01000   |    0.00029    |
|    wrist_3_joint    | origin_yaw |   0.03000   |    0.03509    |
|       Averages      |    ---     |   0.02733   |    0.00677    |
+---------------------+------------+-------------+---------------+

....
```

> Added a crash protection that throws a warning, seen here on the first line

Regarding the terminal screenshot, it finally came time for the student to correct the master aahah :)
https://github.com/lardemua/atom/issues/843#issuecomment-1945841067

I'm closing the issue. I still have this doubt but I think the bug is over.

> This raises another doubt for me... What is the difference between the 2 approaches? When should one define the pattern in the xacro or the world?

closed 2024-03-05 by brunofavs

miguelriemoliveira, 2024-03-05:
> This raises another doubt for me... What is the difference between the 2 approaches? When should one define the pattern in the xacro or the world?

One difference is the one in this issue. One approach has the pattern pose as ground truth, the other does not.

Another difference is that for some systems where the pattern moves around its not possible to put it in the xacro.

brunofavs, 2024-03-05:
Ah ok yeah whenever the pattern is moving it cant be described in the xacro thats right 

Thanks