lardemua/atom #856: Adding a new pattern- ArucoBoard

Ruthrash, 2024-02-29:
Hey! we have  an aruco board pattern already printed out in our lab and we are trying to use atom to calibrated our multimodal sensor system in our robot. I am a bit unsure where all do I have to incorporate changes for this new pattern as I see this being affected in many scripts like test_rgb_labeller, patterns.py, rgb_labeler, data_collector etc as far as I know. Could you please advise what would be the best approach to add this feature?

miguelriemoliveira, 2024-03-01:
Hi @Ruthrash ,

thanks for the interest in ATOM. We currently only support charuco or checkerboards.
I have never used and aruco board, but I think its not compatible with a charuco.

We can add this functionality but it will take a while to work on.

My advise is that you purchase or build a charuco pattern.

Ruthrash, 2024-03-13:
Thanks Miguel! Since we just purchased a new arucoboard pattern, we didn't opt buying a new charuco pattern. However, I was able to add the arucoboard pattern to ATOM codebase. I am testing it right now with our mutli-modal robot system. I can submit a PR after I'm done testing if you think it'd add value to your codebase. Closing this issue now.

closed 2024-03-13 by Ruthrash

miguelriemoliveira, 2024-03-13:
> Thanks Miguel! Since we just purchased a new arucoboard pattern, we didn't opt buying a new charuco pattern. However, I was able to add the arucoboard pattern to ATOM codebase. I am testing it right [...]

That would be great. Thanks!