lardemua/atom #860: riwmpbot real calibration

miguelriemoliveira, 2024-03-02:
Related to #857.

miguelriemoliveira, 2024-03-05:
List of tasks:

- [ ] Evaluation procedure (#857)
- [ ] batch calibrations
- [ ] Create table with reprojection error for experiments, u,f,h,u+f,u+h, f+h, all, and columns erproj sim and real.

miguelriemoliveira, 2024-03-06:
@Kazadhum Created a merged atom dataset containing all collections from 3 datasets, train_diogo, train_miguel and train_bruno.

@manuelgitgomes , I used my script which seems to validate it. If you improve it with the PR please test to make sure it works.

Here's the report on the merged dataset

```bash
Loaded dataset containing 1 sensors and 63 collections.
Dataset contains 1 sensors: ['rgb_world']
Dataset contains 3 patterns: ['forearm_pattern', 'hand_pattern', 'upper_arm_pattern']
Selected collection key is 000
Complete collections (33):['003', '007', '012', '013', '014', '015', '016', '017', '018', '019', '020', '021', '022', '024', '025', '027', '028', '029', '030', '031', '032', '034', '035', '044', '045', '047', '048', '053', '054', '055', '057', '058', '062']
Incomplete collections (30):['000', '001', '002', '004', '005', '006', '008', '009', '010', '011', '023', '026', '033', '036', '037', '038', '039', '040', '041', '042', '043', '046', '049', '050', '051', '052', '056', '059', '060', '061']
Sensor rgb_world has 1 complete detections: ['030']
Sensor rgb_world has 62 partial detections: ['000', '001', '002', '003', '004', '005', '006', '007', '008', '009', '010', '011', '012', '013', '014', '015', '016', '017', '018', '019', '020', '021', '022', '023', '024', '025', '026', '027', '028', '029', '031', '032', '033', '034', '035', '036', '037', '038', '039', '040', '041', '042', '043', '044', '045', '046', '047', '048', '049', '050', '051', '052', '053', '054', '055', '056', '057', '058', '059', '060', '061', '062']

Analysis for pattern forearm_pattern

Collections
+------------+-------------+--------------+
| Collection | is complete |  rgb_world   |
+------------+-------------+--------------+
|    000     |      no     | not detected |
|    001     |      no     | not detected |
|    002     |     yes     |   detected   |
|    003     |     yes     |   detected   |
|    004     |     yes     |   detected   |
|    005     |     yes     |   detected   |
|    006     |      no     | not detected |
|    007     |     yes     |   detected   |
|    008     |      no     | not detected |
|    009     |      no     | not detected |
|    010     |     yes     |   detected   |
|    011     |      no     | not detected |
|    012     |     yes     |   detected   |
|    013     |     yes     |   detected   |
|    014     |     yes     |   detected   |
|    015     |     yes     |   detected   |
|    016     |     yes     |   detected   |
|    017     |     yes     |   detected   |
|    018     |     yes     |   detected   |
|    019     |     yes     |   detected   |
|    020     |     yes     |   detected   |
|    021     |     yes     |   detected   |
|    022     |     yes     |   detected   |
|    023     |     yes     |   detected   |
|    024     |     yes     |   detected   |
|    025     |     yes     |   detected   |
|    026     |      no     | not detected |
|    027     |     yes     |   detected   |
|    028     |     yes     |   detected   |
|    029     |     yes     |   detected   |
|    030     |     yes     |   detected   |
|    031     |     yes     |   detected   |
|    032     |     yes     |   detected   |
|    033     |     yes     |   detected   |
|    034     |     yes     |   detected   |
|    035     |     yes     |   detected   |
|    036     |      no     | not detected |
|    037     |      no     | not detected |
|    038     |      no     | not detected |
|    039     |      no     | not detected |
|    040     |      no     | not detected |
|    041     |     yes     |   detected   |
|    042     |     yes     |   detected   |
|    043     |      no     | not detected |
|    044     |     yes     |   detected   |
|    045     |     yes     |   detected   |
|    046     |     yes     |   detected   |
|    047     |     yes     |   detected   |
[... 147 lines cut ...]
|    053     |     yes     |   detected   |
|    054     |     yes     |   detected   |
|    055     |     yes     |   detected   |
|    056     |      no     | not detected |
|    057     |     yes     |   detected   |
|    058     |     yes     |   detected   |
|    059     |     yes     |   detected   |
|    060     |      no     | not detected |
|    061     |     yes     |   detected   |
|    062     |     yes     |   detected   |
+------------+-------------+--------------+

Joints
+---------------------+----------+---------+----------+----------+-------------+
|        Joint        |   Mean   |   Std   |   Min    |   Max    | Range [deg] |
+---------------------+----------+---------+----------+----------+-------------+
|     elbow_joint     | 1.40869  | 0.17239 | 1.80295  | 1.02612  |   44.50936  |
| shoulder_lift_joint | -1.61119 | 0.23595 | -1.34456 | -2.06688 |   41.38594  |
|  shoulder_pan_joint | 1.41251  | 0.23474 | 1.86586  | 0.76526  |   63.06019  |
|    wrist_1_joint    | -0.79088 | 0.32795 | -0.21639 | -1.59751 |   79.13217  |
|    wrist_2_joint    | -1.85659 | 0.15183 | -1.59865 | -2.37719 |   44.60685  |
|    wrist_3_joint    | 3.06036  | 0.24119 | 3.57142  | 2.44540  |   64.51619  |
|       Averages      | 0.27048  | 0.22734 | 0.68011  | -0.30080 |   56.20178  |
+---------------------+----------+---------+----------+----------+-------------+
```

miguelriemoliveira, 2024-03-06:
Calibrating with the complete merged dataset. 

In the future we should split it into train and test but now were solving and issue with split (#867)

Removed collections that showed high errors (we should have an automatic way to filter these out).
Command is:

    clear && rosrun atom_calibration calibrate  -json ~/datasets/riwmpbot_real/merged/dataset.json -uic  \
    -v -jsf "lambda x: True"  \
    -csf "lambda x: int(x) not in [11,27,32, 34,45,57]"

Initial results look good:

```
Optimization finished in 38.05777 secs: `ftol` termination condition is satisfied.
Errors per collection (anchored sensor,  max error per sensor, not detected as "---")
+------------+----------------+
| Collection | rgb_world [px] |
+------------+----------------+
|    000     |     1.1161     |
|    001     |     1.1461     |
|    002     |     0.6119     |
|    003     |     0.9037     |
|    004     |     1.7056     |
|    005     |     1.9432     |
|    006     |     0.6170     |
|    007     |     1.1882     |
|    008     |     0.9695     |
|    009     |     1.0035     |
|    010     |     1.0006     |
|    012     |     1.1718     |
|    013     |     1.4074     |
|    014     |     1.1588     |
|    015     |     1.0221     |
|    016     |     0.9750     |
|    017     |     1.3319     |
|    018     |     1.1557     |
|    019     |     1.1122     |
|    020     |     1.0657     |
|    021     |     1.2004     |
|    022     |     1.4952     |
|    023     |     1.0941     |
|    024     |     0.7261     |
|    025     |     7.2750     |
|    026     |     1.2092     |
|    028     |     0.7484     |
|    029     |     4.4405     |
|    030     |     0.9349     |
|    031     |     0.7746     |
|    033     |     0.7908     |
|    035     |     0.8720     |
|    036     |     0.8757     |
|    037     |     0.9417     |
|    038     |     0.7274     |
|    039     |     1.0881     |
|    040     |     0.7100     |
|    041     |     0.6608     |
|    042     |     1.0512     |
|    043     |     0.4939     |
|    044     |     0.5334     |
|    046     |     1.1032     |
|    047     |     0.9999     |
|    048     |     0.7095     |
|    049     |     1.3803     |
|    050     |     0.4247     |
|    051     |     0.5064     |
|    052     |     0.7969     |
|    053     |     0.7875     |
|    054     |     0.6085     |
|    055     |     0.5915     |
|    056     |     0.8714     |
|    058     |     0.8515     |
|    059     |     1.1630     |
|    060     |     0.7626     |
|    061     |     0.7227     |
|    062     |     0.6849     |
|  Averages  |     1.1266     |
+------------+----------------+
```

miguelriemoliveira, 2024-03-07:
We should use this command to put the result of the single_rgb_evaluation in the correct place.

`rosrun atom_evaluation single_rgb_evaluation -train_json $ATOM_DATASETS/riwmpbot_real/merged/results/atom_calibration.json -test_json $ATOM_DATASETS/riwmpbot_real/merged/results/atom_calibration.json  -sfr -sfrn $ATOM_DATASETS/riwmpbot_real/merged/results/single_rgb_evaluation.csv`

miguelriemoliveira, 2024-03-08:
#870 is fixed, we can proceed.

For reference, to run the evaluation we should use:

`clear && rosrun atom_evaluation single_rgb_evaluation -train_json $ATOM_DATASETS/riwmpbot_real/merged/atom_calibration.json -test_json $ATOM_DATASETS/riwmpbot_real/merged/dataset.json -csf "lambda x: int(x) not in [11,27,32,34,45,57]" -cpt`

Kazadhum, 2024-03-09:
Running batch calibrations now...

Kazadhum, 2024-03-09:
Batch calibrations run! You can check the results in the drive folder @miguelriemoliveira, but they look good to me!

For reference, here's the case where the joints are not calibrated, with all the patterns used:

```
|----|------------|--------------------------------|-----------------------------|----------------------------------|--------------------|
|    | Collection | rgb_world-forearm_pattern [px] | rgb_world-hand_pattern [px] | rgb_world-upper_arm_pattern [px] | all [px]           |
|----|------------|--------------------------------|-----------------------------|----------------------------------|--------------------|

[.....]

|----|------------|--------------------------------|-----------------------------|----------------------------------|--------------------|
| 12 | Averages   |                        0.80366 |          1.4587800000000002 |               0.6138399999999999 |            1.06582 |
|----|------------|--------------------------------|-----------------------------|----------------------------------|--------------------|

```

And the case where the joints are calibrated, also with all the patterns used:

```
|----|------------|--------------------------------|-----------------------------|----------------------------------|--------------------|
|    | Collection | rgb_world-forearm_pattern [px] | rgb_world-hand_pattern [px] | rgb_world-upper_arm_pattern [px] | all [px]           |
|----|------------|--------------------------------|-----------------------------|----------------------------------|--------------------|

[.....]

|----|------------|--------------------------------|-----------------------------|----------------------------------|--------------------|
| 12 | Averages   |                        0.85301 |                     1.35165 |               0.6880200000000001 |             1.0252 |
|----|------------|--------------------------------|-----------------------------|----------------------------------|--------------------|

```

The results are even better when the joints are calibrated! I think we can make this comparison since I used the same sample seeds for each experiment, so the test/train dataset split is the same between these cases.

miguelriemoliveira, 2024-03-09:
> The results are even better when the joints are calibrated! I think we can make this comparison since I used the same sample seeds for each experiment, so the test/train dataset split is the same between these cases.

Right. That's exactly the same data, with or without joints. That's a good thing.

Results look nice. My only reserve is that the difference between using joints and no joints is quite small 1.0252  to  1.06582.

They may argue its not significant. This was small difference was expected from the sample tests we did, but if we could somehow improve it it would be great.

For that I see several alternatives:

1. Experiment with changing the discarded collections and see if the difference increases
2. Change the balance of the leave-one-out cross validation from 80-20 to 70-30
3. Run a calibration of all joint parameters (not just the yaw) to see if that makes it significantly better

What do you think @Kazadhum ? Any other avenues?

miguelriemoliveira, 2024-03-09:
Another idea is to use the evaluation always with all the patterns so we (the all approach) does not get a worse evaluation when compared to the forearm for example. Its the only way we will have apples to apples comparison.

Kazadhum, 2024-03-10:
> For that I see several alternatives:
> 
>     1. Experiment with changing the discarded collections and see if the difference increases
> 
>     2. Change the balance of the leave-one-out cross validation from 80-20 to 70-30
> 
>     3. Run a calibration of all joint parameters (not just the yaw) to see if that makes it significantly better
> 
> Another idea is to use the evaluation always with all the patterns so we (the all approach) does not get a worse evaluation when compared to the forearm for example. Its the only way we will have apples to apples comparison.

Agreed, these all sound good and in fact, I've run some experiments already. I'll make a TO-DO list here in regards to the different batch experiments we can run, and their results.

- [ ] Experiment with changing the discarded collections and see if the difference increases;
- [ ] Change the balance of the leave-one-out cross validation from 80-20 to 70-30
- [ ] Run a calibration of all joint parameters (not just the yaw) to see if that makes it significantly better

miguelriemoliveira, 2024-03-13:
Great, looking forward to see them.

Kazadhum, 2024-03-13:
Here are the results, @miguelriemoliveira ...

## Using all the patterns, calibrating **without** joints:

|    | Collection | rgb_world-forearm_pattern [px] | rgb_world-hand_pattern [px] | rgb_world-upper_arm_pattern [px] | all [px]           |
|----|------------|--------------------------------|-----------------------------|----------------------------------|--------------------|
| 11 | Averages   |             0.9443199999999999 |          1.2479599999999997 |               0.6427299999999999 | 1.0081599999999997 |

## Using all the patterns, calibrating **only with the yaw join parameter**:

|    | Collection | rgb_world-forearm_pattern [px] | rgb_world-hand_pattern [px] | rgb_world-upper_arm_pattern [px] | all [px]           |
|----|------------|--------------------------------|-----------------------------|----------------------------------|--------------------|
| 11 | Averages   |             0.8950099999999999 |                     1.09946 |               0.6506199999999999 |             0.9172 |

## Using all the patterns, calibrating **with all the joint parameters**:

|    | Collection | rgb_world-forearm_pattern [px] | rgb_world-hand_pattern [px] | rgb_world-upper_arm_pattern [px] | all [px]           |
|----|------------|--------------------------------|-----------------------------|----------------------------------|--------------------|
| 11 | Averages   |                        0.63278 |          1.0400800000000001 |                          0.52929 | 0.8022599999999999 |

miguelriemoliveira, 2024-03-13:
Houston, we have a paper?

Kazadhum, 2024-03-13:
I believe so! Onwards to making tables!

Kazadhum, 2024-03-14:
Hey @miguelriemoliveira! It might also be cool to point out that **even when we use only the hand pattern**, the calibration has better results when the joints are included in the optimization!

Note, however, that in order to prove that point, we need to look to the error value related to the hand pattern instead of the 'all' value, since we're only using that pattern for the calibration.

I don't know if it's clear what I meant, but basically I think it might be good to present these results as support for the idea that:
1. Using multiple patterns is beneficial, regardless of optimizing the joint parameters;
2. Optimizing the joint parameters is not only doable but also impacts the calibration results significantly, regardless of if we only use the hand pattern or if we use all the patterns;
3. Doing both of these things is even better.

miguelriemoliveira, 2024-03-14:
> I don't know if it's clear what I meant, but basically I think it might be good to present these results as support for the idea that:

I think so. You are proposing that we evaluate no joints vs joints yaw vs joints all without having the element of multiple patterns, using only the hand pattern and the evaluation only with the hand pattern.

> Using multiple patterns is beneficial, regardless of optimizing the joint parameters;
> Optimizing the joint parameters is not only doable but also impacts the calibration results significantly, regardless of if we only use the hand pattern or if we use all the patterns;
> Doing both of these things is even better.

Right, we propose two improvements which are not coupled. They both  independently enhance the calibration, right?

miguelriemoliveira, 2024-03-14:
Let's meet to decide the tables for the paper?

Kazadhum, 2024-03-14:
Hi @miguelriemoliveira !

> Right, we propose two improvements which are not coupled. They both independently enhance the calibration, right?

Exactly.

> Let's meet to decide the tables for the paper?

Sounds good! I can meet now until around 18h. Afterwards, I can meet after dinner.

miguelriemoliveira, 2024-03-28:
We can close this, right. These tables are complete, all we're missing are the comparisons to others, right?

closed 2024-03-28 by miguelriemoliveira

Kazadhum, 2024-03-28:
Hi @miguelriemoliveira ! Yes, I think so!