lardemua/atom #861: Create script to merge datasets

miguelriemoliveira, 2024-03-02:
Hi @manuelgitgomes ,

I know you did a dataset split script . How  easy  would it be for you to do a merge?

Thanks

miguelriemoliveira, 2024-03-05:
Thanks @manuelgitgomes. Will test it now.

miguelriemoliveira, 2024-03-05:
wait, @manuelgitgomes this is the split. The questions was about a merge tool ...

miguelriemoliveira, 2024-03-05:
I was trying to implement but now that I think of it, perhaps its easier to merge the bagfiles and then create a single datasets.

miguelriemoliveira, 2024-03-05:
workign on it ...

miguelriemoliveira, 2024-03-05:
I think I got a first solution up and running ... will test later.

manuelgitgomes, 2024-03-05:
Sorry, I have create a new PR #864 to solve this in a more general way (I hope so)

miguelriemoliveira, 2024-03-05:
Hi @manuelgitgomes ,

but I also did it. I said above "working on it" to let you know not to worry about this.

miguelriemoliveira, 2024-03-16:
Closing this one. The version we have is functional.

closed 2024-03-16 by miguelriemoliveira