lardemua/atom #865: riwmpbot only has data regarding one joint in each collection

Kazadhum, 2024-03-06:
Hi @miguelriemoliveira ! Each collection in the dataset for riwmpbot in atom_examples only has the data regarding one joint. Maybe the file needs to be updated?

miguelriemoliveira, 2024-03-06:
This occurs probably when the dataset was produced using  a calibration configuration where only the elbow joint was being optimized.

In any case I downloaded the train.zip  and I think  we have all the joints ...

miguelriemoliveira, 2024-03-06:
How are you downloading the dataset?

miguelriemoliveira, 2024-03-06:
The link should be 

https://drive.google.com/file/d/1bzgNoFnqLvJ0zE8bvWJjVV50w2bfkage/view?usp=sharing

Kazadhum, 2024-03-06:
Hi! I downloaded the dataset from the link given here: https://github.com/lardemua/atom/blob/noetic-devel/atom_examples/riwmpbot/README.md#collecting-a-dataset

If you check the link there with the link you've sent, you can see they're different. The dataset you've just sent seems correct.

Kazadhum, 2024-03-06:
Ah I see. I'll update this one then, for anyone who tries to do the same as I did, but I'll do as you suggest in the future. Thank you!

Kazadhum, 2024-03-06:
Done!

closed 2024-03-06 by Kazadhum