lardemua/atom #880: Replicate the softbot calibration

miguelriemoliveira, 2024-03-14:


brunofavs, 2024-03-15:
Renaming the issue, will use this issue to document the progress in creating the atom example for softbot

brunofavs, 2024-03-15:
I recorded a dataset and for some reason during the calibration the pattern is facing backwards and Im having huge projection errors.

[image]

```
Errors per collection (anchored sensor,  max error per sensor, not detected as "---")
+------------+------------------------+-------------------------+
| Collection | front_left_camera [px] | front_right_camera [px] |
+------------+------------------------+-------------------------+
|    001     |        32.3592         |         39.7152         |
|    002     |        46.8338         |         38.4344         |
|    003     |        24.1222         |         26.5183         |
|    004     |        32.3592         |         39.5004         |
|  Averages  |        33.9186         |         36.0421         |
+------------+------------------------+-------------------------+
```

I'm unsure why this is happening.

I checked if the pattern definition in the config file matches the shown pattern

brunofavs, 2024-03-15:
@manuelgitgomes told me to use the updated pattern mesh in atom, I'm using a old version in softbot.

brunofavs, 2024-03-15:
Previously the mesh file was being stored locally:

` "package://softbot_gazebo/models/charuco_800x600/charuco_800x600.dae"`

I'm updating it to use atom's version as atom is already a dependency anyway.

` "package://atom_worlds/pattern/models/charuco_800x600_5x5_100/charuco_800x600_5x5_100.dae"`

However during this I realized the pattern was supposed to be "fixed", and I moved the pattern with the interactive marker whie recording the bagfile to make it easier.

I will try to change in the dataset.json the keyword to false and see if it works better. Otherwise I will record a new bagfile.

Perhaps if the pattern is "fixed", the interactive pattern shouldn't launch? Would still be useful if the user wanted to move the pattern to a new location before starting to record.

miguelriemoliveira, 2024-03-15:
> However during this I realized the pattern was supposed to be "fixed", and I moved the pattern with the interactive marker whie recording the bagfile to make it easier.

Fixed means the transformation between the pattern link and the pattern's parent link does not change over all collections.

You could, theoretically, start the bagfile, move the pattern to some pose, then take all collections with that pose.

However, this is an atom example, and for these the standards are high. You should have a bagfile where the pattern does not move.

> I will try to change in the dataset.json the keyword to false and see if it works better. Otherwise I will record a new bagfile.

Not a good idea is my opinion.

> Perhaps if the pattern is "fixed", the interactive pattern shouldn't launch? Would still be useful if the user wanted to move the pattern to a new location before starting to record.

Well, sometimes you want to do what I said above and you suggest here. But for the examples start the bag with the pattern in the correct place.

miguelriemoliveira, 2024-03-15:
We can try to zoom later ... are you available?

brunofavs, 2024-03-15:
Yes Im available at any time :)

I didnt answer to your comment yet because Im not back on the computer

brunofavs, 2024-03-15:
> Fixed means the transformation between the pattern link and the pattern's parent link does not change over all collections.

Actually this is helpful because I thought it was fixed from world to the pattern link.

> You could, theoretically, start the bagfile, move the pattern to some pose, then take all collections with that pose.

Yes, maybe my wording was confusing but was what I was also thinking here:

> Would still be useful if the user wanted to move the pattern to a new location before starting to record.

**Regarding just changing the keyword in the dataset :**

> Not a good idea is my opinion.

It did work out, with subpixel accuracy, but I will redo it again with the static pattern to make the atom example as simple as possible. For the same reason I decided to exclude the lidar from the calibration to focus this example on atom's capability to calibrate system's whose  tf's are not totally defined by the robot state publisher, as is the case here with the moving robot.

```

Errors per collection (anchored sensor,  max error per sensor, not detected as "---")
+------------+------------------------+-------------------------+
| Collection | front_left_camera [px] | front_right_camera [px] |
+------------+------------------------+-------------------------+
|    001     |         1.0605         |          0.9325         |
|    002     |         0.3494         |          0.4292         |
|    003     |         0.4171         |          0.1881         |
|    004     |         0.8091         |          0.7295         |
|  Averages  |         0.6590         |          0.5698         |
+------------+------------------------+-------------------------+
```

brunofavs, 2024-03-18:
I don't know why but the pattern is again showing backwards

[image]

I'm confused, it seemed to get fixed here https://github.com/lardemua/atom/issues/880#issuecomment-1999914243 , but now its back again even though I'm using atom's models instead of softbot's :C

Initially it was not even showing any pattern, I changed the softbot.world to use the atom's models as I deleted the outdated softbot's ones and it started showing again but  backwards

brunofavs, 2024-03-18:
Using atom's calibration_studio.world fixes the problem, seems to be a very similar world I'll use this one instead.

The problem lies somewhere in softbot.world

brunofavs, 2024-03-18:
For some reason, my pattern isn't loading in rviz.

It happens both in the bringup.launch and collect_data.launch

[image]

It isn't easy to see if the LiDARs are aligned with the pattern without seeing the pattern.

brunofavs, 2024-03-18:
Updates : 

Struggled a lot with LiDARs completely out of position. Experimented with a lot of stuff like anchoring sensors, removing collections ... After talking with Manuel he told me to verify I didn't mislabeled any 3D points by accident. That turned out to be the culprit, that was the motivation for creating #887.

Now I have the calibration working properly, with decent results for an initial noisy guess of 2cm and 2º (`-nig 0.02 0.02`)
```
+------------+------------------------+-------------------------+-------------+
| Collection | front_left_camera [px] | front_right_camera [px] | lidar3d [m] |
+------------+------------------------+-------------------------+-------------+
|    001     |         1.1104         |          0.8573         |    0.0059   |
|    002     |         0.9173         |          0.7258         |    0.0068   |
|    003     |         0.7595         |          0.5824         |    0.0067   |
|    004     |         0.4805         |          0.4594         |    0.0053   |
|  Averages  |         0.8169         |          0.6562         |    0.0062   |
+------------+------------------------+-------------------------+-------------+
```

[image]

Finally, after battling all sorts of miscellaneous bugs, I have every piece of data I need to write softbot's example and move on to the next task.

miguelriemoliveira, 2024-03-18:
> Using atom's calibration_studio.world fixes the problem, seems to be a very similar world I'll use this one instead.
> 
> The problem lies somewhere in softbot.world

Yeah. Might as well remove that one.

miguelriemoliveira, 2024-03-18:
> After talking with Manuel he told me to verify I didn't mislabel any 3D points by accident.

But I told to check the labels using the dataset playback on Friday ...

miguelriemoliveira, 2024-03-18:
> that was the motivation for creating #887.

I'll take a look and reply there.

miguelriemoliveira, 2024-03-18:
> Now I have the calibration working properly, with decent results for an initial noisy guess of 2cm and 2º (

Looks nice. Congrats.

miguelriemoliveira, 2024-03-18:
> Finally, after battling all sorts of miscellaneous bugs, I

Welcome to ATOM!

manuelgitgomes, 2024-03-19:
Hello to both,

Nice work @brunofavs!

It would be nicer to have the patterns closer to the ground, so that the LiDARs detect more of the pattern!

brunofavs, 2024-03-19:
> > After talking with Manuel he told me to verify I didn't mislabel any 3D points by accident.
> 
> But I told to check the labels using the dataset playback on Friday ...

Yeah I remember and I was careful but not careful enough it seems, there was a line of really tiny far away points that I didn't see.

> > Using atom's calibration_studio.world fixes the problem, seems to be a very similar world I'll use this one instead.
> > The problem lies somewhere in softbot.world
> 
> Yeah. Might as well remove that one.

Ok I will delete that one then.

brunofavs, 2024-03-19:
> Hello to both,
> 
> Nice work @brunofavs!
> 
> It would be nicer to have the patterns closer to the ground so that the LiDARs detect more of the pattern!

Yep you are absolutely right, I thought the LiDAR would reach higher than it did, I will keep that in mind from now on :)

miguelriemoliveira, 2024-03-19:
So let us know when you have a PR. I would like to test the calibration on my side.

Also, for the readme, you probably want to add links to bags and datasets. Send them to me and I will give you the link.

brunofavs, 2024-03-19:
Yes Im writting the readme right now.
I will send a PR when I have it ready.

Should be finished today if everything goes well. I was battling #888 for a good hour today.
I will send the bags and datasets here in a moment.

brunofavs, 2024-03-19:
https://we.tl/t-McfWJeAEYt

Here it is @miguelriemoliveira

brunofavs, 2024-03-20:
Hey,

I finished the initial draft for the softbot example. Its missing the links for the bagfiles/datasets.

https://github.com/miguelriemoliveira/softbot/blob/b3223e9bc98d2eb3d58ea340e7cfba375ad06bef/README_Example.md

I'd like to get some feedback.

The full evaluation gave me a lot of different tables and I'm not sure if the ones I put in the example are enough.

Also the LiDAR evaluation crashed I dont know why..

```
Traceback (most recent call last):
  File "/home/brunofavs/atom_ws/src/atom/atom_evaluation/scripts/lidar_to_rgb_evaluation", line 168, in <module>
    pts_in_image = rangeToImage(collection, args['test_json_file'], pattern_key,
TypeError: rangeToImage() takes 5 positional arguments but 6 were given
INFO - 2024-03-20 17:58:33,818 - core - signal_shutdown [atexit]
Traceback (most recent call last):
  File "/home/brunofavs/atom_ws/src/atom/atom_evaluation/scripts/inter_collection_lidar_to_rgb_evaluation", line 180, in <module>
    pts_in_image = rangeToImage(collection_target, args['test_json_file'], pattern_key,
TypeError: rangeToImage() takes 5 positional arguments but 6 were given
INFO - 2024-03-20 17:58:33,882 - core - signal_shutdown [atexit]
Traceback (most recent call last):
  File "/home/brunofavs/atom_ws/src/atom/atom_evaluation/scripts/lidar_to_rgb_evaluation", line 168, in <module>
    pts_in_image = rangeToImage(collection, args['test_json_file'], pattern_key,
TypeError: rangeToImage() takes 5 positional arguments but 6 were given
INFO - 2024-03-20 17:58:33,887 - core - signal_shutdown [atexit]
Traceback (most recent call last):
  File "/home/brunofavs/atom_ws/src/atom/atom_evaluation/scripts/inter_collection_lidar_to_rgb_evaluation", line 180, in <module>
    pts_in_image = rangeToImage(collection_target, args['test_json_file'], pattern_key,
TypeError: rangeToImage() takes 5 positional arguments but 6 were given
INFO - 2024-03-20 17:58:33,930 - core - signal_shutdown [atexit]
```

I honestly didn't look into it a lot, but seems like someone changed the rangeToImage function sometime ago and forgot to update it there.

Do you have time to meet tomorrow @miguelriemoliveira ? Or today I'm free too but it might be too late already

brunofavs, 2024-03-20:
@manuelgitgomes Tagging in case you want to join too when we meet.

miguelriemoliveira, 2024-03-22:
Hi @brunofavs ,

sorry for the delay. Sometimes I cannot answer immediately. We can meet today at 15h30?

brunofavs, 2024-03-22:
Hey @miguelriemoliveira 
Yes I'm available to meet at that time :)

I mentioned it on the other issue because I was worried you didn't see it by accident because I wrote on many issues many things the past few days and it could've been lost in the crowd.

miguelriemoliveira, 2024-03-22:
Looks great. Please create a PR.

brunofavs, 2024-03-22:
Yes once I patch up the little things we discussed in the meeting I will :)

miguelriemoliveira, 2024-03-28:
Hi @brunofavs ,

I did not receive the PR for this ...

brunofavs, 2024-03-28:
Hey @miguelriemoliveira 
I had a few hiccups the past few days, it's almost finished.

Today I will push the PR

brunofavs, 2024-03-28:
PR https://github.com/lardemua/atom/pull/907

miguelriemoliveira, 2024-03-29:
Great work. Thanks.

closed 2024-03-29 by miguelriemoliveira