lardemua/atom #894: collect real data at INESC 2024 (zaubot)

miguelriemoliveira, 2024-03-22:


miguelriemoliveira, 2024-03-27:
Hi all,

please create issues for each task above and add the number, like in the example above.

Kazadhum, 2024-03-28:
Hello @manuelgitgomes and @brunofavs! I talked with @miguelriemoliveira and we had the idea of gathering on the 8th of April at LAR to try and get all the sensors working together.

Are you available?

brunofavs, 2024-03-28:
Hey
The proposed date sounds good to me :)

miguelriemoliveira, 2024-04-03:
Hello to all. Thank you for your efforts in making each sensor work separately. I think that milestone is almost complete. 
One thing we have to have tough, is the ability to change the resolution of the sensors (perhaps not the velodyne) and the data frequency (including the velodyne). 

Ideally each launch file for the sensor should have arguments where we can define these parameters.

I am saying this because I am anticipating a very large data throughput, which can be solved by reducing resolution and freqquency.

miguelriemoliveira, 2024-04-03:
I think so. Because we're using the bagfile for calibration then we could use just 5 or 10 hz. If that is not enough we can reduce resolution of cameras

manuelgitgomes, 2024-04-30:
Data collection done, closing this.

closed 2024-04-30 by manuelgitgomes