lardemua/atom #899: Enhance cross validation

manuelgitgomes, 2024-03-26:
Enhance cross validation by using scikit-learn functions.

manuelgitgomes, 2024-04-24:
Hello @miguelriemoliveira  and @Kazadhum.

I have now added cross validation support in batch execution on the branch `dev/cross-validation`.

The user now defines type of cross validation and its parameters in `data.yaml`:
https://github.com/lardemua/atom/blob/b61df4d9019f76c921ac605441ca162c20c77e50/atom_batch_execution/experiments/rrbot_example/data.yml#L23-L26

Right now, fold creation is supported with StratifiedKFold, KFold, LeaveOneOut, and StratifiedShuffleSplit from scikit-learn.
The classes used for stratification are the combination of sensors and patterns detected, as written in:
https://github.com/lardemua/atom/blob/b61df4d9019f76c921ac605441ca162c20c77e50/atom_batch_execution/scripts/batch_execution#L30-L54

This then creates a new `auto_rendered.yaml`, with division of each run in folds, using `-csf` to define the collections used by the fold:

https://github.com/lardemua/atom/blob/b61df4d9019f76c921ac605441ca162c20c77e50/atom_batch_execution/experiments/rrbot_example/auto_rendered.yaml#L27-L44

`process_results` is also adapted to run with these folds!

I have done a test with rrbot and everything seemed nice, can you test on your machines?

miguelriemoliveira, 2024-04-25:
Hi @manuelgitgomes ,

looks great. Thanks. 

@Kazadhum I do not have a lot of time right now. Can you test it please?

One question: if we want the run the old way, is it possible or not?

manuelgitgomes, 2024-04-26:
> One question: if we want the run the old way, is it possible or not?

Right now, you can't, but I can change that easily enough (I think). On it.

> BTW, is it still the case that it doesn't make sense to have the `collection` column in the processed results, since it also averages the collection number?

It doesn't, I believe. I can delete it.

Kazadhum, 2024-04-26:
> > BTW, is it still the case that it doesn't make sense to have the `collection` column in the processed results, since it also averages the collection number?
> 
> It doesn't, I believe. I can delete it.

Now that I mention it, I think it doesn't make much sense to have any lines at all besides the "Average" line, since all other values are "meaningless" (as they belong to different collections). Do you agree?

manuelgitgomes, 2024-04-26:
> Now that I mention it, I think it doesn't make much sense to have any lines at all besides the "Average" line, since all other values are "meaningless" (as they belong to different collections). Do you agree?

In the processed results? Sure, makes sense. Can also be removed, but I don't see any wrong in leaving it there, right?

miguelriemoliveira, 2024-04-26:
Looks great!

manuelgitgomes, 2024-04-30:
This has been merged to main, closing.

closed 2024-04-30 by manuelgitgomes