lardemua/atom #903: Implement Intel RealSense L515 to prepare for data collection at INESC 2024 (zaubot)

manuelgitgomes, 2024-03-28:
This is done! I implemented a launch file to launch this camera and instructions on how to use are on the [README.md](https://github.com/lardemua/zau/blob/main/README.md).
There was only one caveat with the USB rules for this sensor, but the instructions to solve it are on the README.md as well.

[image]

miguelriemoliveira, 2024-03-28:
Hi @manuelgitgomes ,

how about controlling the publication frequency and camera's resolution? Is it possible? What are the current values?

manuelgitgomes, 2024-03-28:
Hello @miguelriemoliveira.

Answering your question, yes, it is changeable but only to predefined resolutions/fps combinations.
The bad news is that I cannot find a list of these combinations.
I tried searching around in librealsense but I could not find anything, but I can still explore more.

I reopened this issue because when trying to answer you questions I found a critical error when using the L515 camera.
In my experience, when launching the camera various times in somewhat quick succession, my OS freezes completely.

I found various people with the same issue [[1]](https://github.com/IntelRealSense/librealsense/issues/10207)[[2]](https://github.com/IntelRealSense/librealsense/issues/10159).
The solution for all involved not installing the binaries of the realsense-ros, but installing by source.
This causes a problem, as it involves patching the kernel in order to support the cameras.
The kernel I am currently running on my host is not supported by the [librealsense](https://github.com/IntelRealSense/librealsense/tree/master). 
My Ubuntu 20.04 docker image shares kernel versions with the host, so it does not work.

I could try and create a docker image running this kernel version.
I can have problems with this as I am uncertain if I need any library on host device to detect the camera, as I needed to change my udev rules on host as well.

We can also try in a computer that has a compatible kernel version.

miguelriemoliveira, 2024-03-28:
Hum, these are not good new. The solution for a new docker with a compatible kernel is not good, because this is just one sensor from a large set of sensors. How about disregarding this sensor and using the asus camera instead as the rgbd sensor?

brunofavs, 2024-03-28:
Might be a long shot but VM's don't share the kernel with the host. Given that you @manuelgitgomes  have KVM sorted in your personal desktop it may be a option to experiment there.

That or we must not launch the camera repeatedly, which is annoying indeed :c

brunofavs, 2024-03-28:
Also I noticed the issues you linked belong to different realsense sensors, I hope those problems don't carry over to the my *T265* as I'm pretty certain we are using the same kernel.

manuelgitgomes, 2024-04-02:
> Might be a long shot but VM's don't share the kernel with the host. Given that you @manuelgitgomes have KVM sorted in your personal desktop it may be a option to experiment there.

Yes, but that is a long-shot, honestly. 
I don't know if it is a feasible solution.

> How about disregarding this sensor and using the asus camera instead as the rgbd sensor?

That could be an option, but:

> I hope those problems don't carry over to the my _T265_ as I'm pretty certain we are using the same kernel.

I am fairly certain these problems will carry out to the T265.
I will continue trying and if I figure something out I'll say it.

brunofavs, 2024-04-03:
Hey.
I tested thoroughly the **L515** on my computer and I couldn't replicate the crash.
Run for long periods of time, restarted the device more than 10 times, all seemed fine.

I'm currently on this kernel `6.6.23-1-lts`

miguelriemoliveira, 2024-04-03:
That's good news.

miguelriemoliveira, 2024-04-03:
I think your laptop is becoming the prime candidate to be the system which will be used to collect all data.

What are your specs?

brunofavs, 2024-04-03:
My pc is the Asus ROG Strix G531GT and the main specs are:
i7-9750h
GTX 1650 
16Gb Ram
3xUSB 3.2 Gen 1.

My main worry would be the battery life. I doubt it lasts even 1 hour with a full ROS system running full blast.
I also have a RJ45 port to connect the LiDAR, so no extra adaptor is necessary.

I don't mind using mine but I reckon your pc @miguelriemoliveira is superior to mine.

miguelriemoliveira, 2024-04-03:
> I don't mind using mine but I reckon your pc @miguelriemoliveira is superior to mine.

right, but I will not be with you three most of the times. I am not even sure I will going to INESC in all the days of the tests.

So best to aim to use your laptop. Battery is not a problem as we can have the charger always connected, even with the agv moving around. 

Tagging @Kazadhum for visibility...

brunofavs, 2024-04-03:
Ok that settles it then. Do you have perhaps a USB hub we can borrow?

I only have 3 USB ports and we will certainly need more. 
I talked with @manuelgitgomes but his hub has USB-c input which I do not have in my laptop.

Kazadhum, 2024-04-03:
Hello @miguelriemoliveira @brunofavs and @manuelgitgomes !

I agree that @brunofavs's PC is most likely the best candidate, especially since it appears to run everything with no problems. I think I have a USB Hub somewhere if needed, but I never used it so I don't even know if it works...

manuelgitgomes, 2024-04-30:
Data collection done, closing this.

closed 2024-04-30 by manuelgitgomes