lardemua/atom #905: Move atom examples data to NAS

miguelriemoliveira, 2024-03-28:


miguelriemoliveira, 2024-04-01:
Hi @manuelgitgomes , @Kazadhum and @brunofavs ,

I was working on this last week. I could not find a way to create a script to automatically download data from the NAS.
The implications are that the script to download the data from all atom examples at once would disapear.

For the atom team we have the solution to setup the synology drive, so that script won't be necessary, but other atom users would have to download the data piece by piece. 

Do you think this is a big shortcoming, or that we may advance like this?

Kazadhum, 2024-04-01:
Hello @miguelriemoliveira! I don't really think it's a big problem.

Trying to imagine being someone trying to learn how to use ATOM, which I assume is the bulk of people using the `atom_examples`, I think I'd go system-by-system and I'd imagine I would consult the documentation for each system anyways. So I don't really think it's such a big inconvenience to download a system at a time...

Let's see what @brunofavs and @manuelgitgomes say...

manuelgitgomes, 2024-04-02:
> Do you think this is a big shortcoming, or that we may advance like this?

I do not think so, and I agree with @Kazadhum.

miguelriemoliveira, 2024-04-02:
On it...

brunofavs, 2024-04-02:
I think all the links for the "downloadables" should be at the top of each example then, instead of spread out on the example readme, often in different places. It would be more systematic.

Another thing to take into consideration is that as is, the user won't have problems with wrong folder structure as its all predefined by the download script. But as long as we have a note here ...
 https://github.com/lardemua/atom/blob/b5b9574ae97d09a561767462de3434a4d6c158a9/atom_examples/README.md?plain=1#L25 

... explaining how the folder structure should look it should be fine.

miguelriemoliveira, 2024-04-02:
Right @brunofavs ,

thank makes sense.

Thanks.