lardemua/atom #908: Replace placeholders with links in softbot example

brunofavs, 2024-03-28:
Creating this issue as @miguelriemoliveira requested me to replace these placeholders in the softbot example : 

It's just here : 

https://github.com/lardemua/atom/blob/6d3e92d734c2a04ce7193fb59d2611e08ca6b3b7/README.md#L77-L83

and here : 

https://github.com/lardemua/atom/blob/6d3e92d734c2a04ce7193fb59d2611e08ca6b3b7/README.md#L116-L123

and here : 

https://github.com/lardemua/atom/blob/6d3e92d734c2a04ce7193fb59d2611e08ca6b3b7/README.md#L180-L187

With this [link](https://we.tl/t-etWHBKnye1)

I do realize I forgot to update the test dataset.
I will do it shortly and comment this issue

brunofavs, 2024-03-28:
Updated link with test and train dataset
https://we.tl/t-ssxiMUx4AV

miguelriemoliveira, 2024-03-29:
Hi @brunofavs ,

here are the links you can use to update the readme:

train bagfile
https://gofile.me/7nlMj/PAmVjWhxy

test bagfile
https://gofile.me/7nlMj/czx2rL51T

train dataset
https://gofile.me/7nlMj/5XvOf6TS2

test dataset
https://gofile.me/7nlMj/P7BflqUVp

Do you need anything else?

closed 2024-04-02 by brunofavs

miguelriemoliveira, 2024-04-02:
Great. Thanks.