lardemua/atom #909: Implement Intel RealSense T265 to prepare for data collection at INESC 2024 (zaubot)

brunofavs, 2024-03-28:
Related to #894

manuelgitgomes, 2024-04-03:
Hello @brunofavs.

I did not enconter such errors, however, I needed to add some udev rules to my host system, as is stated in the repo:
https://github.com/lardemua/zau

brunofavs, 2024-04-03:
Ok , I added those to the docker system instead with no luck. Will try in the host system now.

brunofavs, 2024-04-03:
Yep that worked. 

https://github.com/lardemua/atom/assets/31831043/eb556136-75ae-4f8e-bfe6-dec5f6c0909d

These are the topics published : 

```
/camera/accel/imu_info
/camera/accel/metadata
/camera/accel/sample
/camera/gyro/imu_info
/camera/gyro/metadata
/camera/gyro/sample
/camera/odom/metadata
/camera/odom/sample
/camera/realsense2_camera_manager/bond
/camera/tracking_module/parameter_descriptions
/camera/tracking_module/parameter_updates
/clicked_point
/diagnostics
/initialpose
/move_base_simple/goal
/rosout
/rosout_agg
/tf
/tf_static
```

I'm wondering why our camera isn't publishing these topics as stated [here](https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#using-t265)
```
/camera/fisheye1/image_raw
/camera/fisheye2/image_raw
```

There is one **Warning** when the sensor moves too fast, which is interesting : 
`WARNING [125497602209536] (tm-device.cpp:1291) SLAM_ERROR Speed`

Tagging @miguelriemoliveira  for visibility.

brunofavs, 2024-04-03:
I'm kinda new to dockers, I just started using Distrobox 1/2 weeks ago. I would say I plan to run everything in the same container but I really have no idea which benefits using multi containers would bring to the table.

But I think testing that wouldn't be hard, I would just need to clone the docker.
Granted I also don't know how to use ROS in a master/slaves configuration.

miguelriemoliveira, 2024-04-03:
> Granted I also don't know how to use ROS in a master/slaves configuration.

That part I can cover, no problem.

I think we should target using a single docker container. That would be the easiest.

brunofavs, 2024-04-03:
Ok, sounds easier to me too.

manuelgitgomes, 2024-04-30:
Data collection done, closing this.

closed 2024-04-30 by manuelgitgomes