lardemua/atom #917: Placement of atom_examples

manuelgitgomes, 2024-04-02:
Hello @miguelriemoliveira, @Kazadhum, @brunofavs 
What do you all think of moving atom_examples to a new repo?

The reason behind this is that atom_examples is now responsible for a large portion of the size of the repository.
Without the .git and the .venv, atom is 164M in size, 129M of those are atom_examples (around 79%):

```
353M	atom/.venv
8.0K	atom/README.md
596K	atom/atom_evaluation
2.0M	atom/atom_batch_execution
36K	atom/LICENSE
448M	atom/.git
129M	atom/atom_examples
500K	atom/atom_core
12K	atom/dockerfiles
30M	atom/docs
1.7M	atom/atom_worlds
1.3M	atom/atom_calibration
12K	atom/.vscode
4.0K	atom/.gitignore
84K	atom/atom_msgs
4.0K	atom/requirements.txt
52K	atom/atom_rviz_plugins
12K	atom/atom
965M	atom
```

I think most ATOM users, outside of the ATOM team, don't use them.
And the ones who do, I do not think it will be a big inconvenience to download a new repo.
This can also come in handy if we release the package in apt in the future.

I may also be reading way too much into this, but feel free to state your opinion.

miguelriemoliveira, 2024-04-02:
Hi @manuelgitgomes ,

not sure. I like the fact that its encapsulated all together, but the space could become a problem.

I went a dig a bit deeper into the space, and this is the result:

[image]

So I think the problem is located in very specific places (e.g. images in docs), which we can solve. 

What do you think?

miguelriemoliveira, 2024-04-02:
I used command 

    mogrify -resize 820x ./*.png

in docs to reduce image sizes. Also, some images were not used. Did this for softbot only.

manuelgitgomes, 2024-04-02:
Makes sense @miguelriemoliveira!