lardemua/atom #918: Commands for a multi RealSense device system integration

brunofavs, 2024-04-03:
This is just a list that could help us integrating the sensors next week when we meet.

`tf_prefix: By default all frame's ids have the same prefix - camera_. This allows changing it per camera.`

`<stream_name>_frame_id, <stream_name>_optical_frame_id, aligned_depth_to_<stream_name>_frame_id: Specify the different frame_id for the different frames. Especially important when using multiple cameras.`

This can also be either by USB port ID or Serial Number

`device_type: will attach to a device whose name includes the given device_type regular expression pattern. Default, ignore device type. For example, device_type:=d435 will match d435 and d435i. device_type=d435(?!i) will match d435 but not d435i.`

`enable_sync: gathers closest frames of different sensors, infra red, color and depth, to be sent with the same timetag. This happens automatically when such filters as pointcloud are enabled.`

brunofavs, 2024-04-03:
You threw me in a big loop transfering the issue to this repo. I legit thought it vanished, I couldn't find it anywhere ahahaah.

> hum, are you now going to create instructions on the repository's readme?

Yes I started it already, but I didn't push yet.

miguelriemoliveira, 2024-04-03:
Ooops, sorry.

brunofavs, 2024-04-03:
Working with L515 and T265 **without usb hub**
https://github.com/lardemua/atom/assets/31831043/61b19ddd-6a48-4b0a-9f0d-e89bd37fcd54

Will try using a USB hub and integrating the asus camera into the mix aswell

brunofavs, 2024-04-03:
> Will try using a USB hub

Works seamlessly as well with a usb hub and the 2 realsense's

miguelriemoliveira, 2024-04-03:
Great news. We're only missing the velodyne then?

brunofavs, 2024-04-03:
And the astras, which are giving me a headache rn, but I will comment on @Kazadhum 's issue if I cant solve them meanwhile

brunofavs, 2024-04-03:
Astra alone working  : 
[image]

Will test with the others now

brunofavs, 2024-04-03:
Now the RGB but no depth.
AAAAAAA

I'm clueless

[image]

miguelriemoliveira, 2024-04-03:
@brunofavs  the asta cameras are very bad. We will not use them for getting depth, only rgb. 

The problem is with their drivers, that reserved the entire usb bandwidth. I lost 1 month with @danifpdra on this several months ago.

So bottom line, astras are rgb only cameras for us. We will not use their drivers, instead we use the usb camera driver from ros.

@Kazadhum did this right?

Kazadhum, 2024-04-03:
Hello @miguelriemoliveira and @brunofavs ! I only used 1 astra at a time when testing last week, so maybe that's a problem that I didn't catch... But from what I understand the issue comes from using two depths... If we only use the RGB cameras (with the usb camera drivers), I imagine you won't run into trouble.

Can you try disabling the depths, @brunofavs?

miguelriemoliveira, 2024-04-03:
> Can you try disabling the depths, @brunofavs?

 Better, like @Kazadhum  says, to use the usb camera drive (which only operates for rgb)

brunofavs, 2024-04-04:
[image]

LiDAR is working. The process was super easy.
I just add to tweak the network commands @manuelgitgomes provided because my ethernet interface is eno2 and not eth1 as his is. I think its because @manuelgitgomes is using an adapter. 
(https://superuser.com/questions/1053003/what-is-the-difference-between-eth1-and-eno1)

brunofavs, 2024-04-04:
> > Can you try disabling the depths, @brunofavs?
> 
> Better, like @Kazadhum says, to use the usb camera drive (which only operates for rgb)

Ok I will do that @miguelriemoliveira . If I'm not mistaken, rn its using the astra drivers for depth and usb-cam for rgb, so I just gotta disable a node in the launch file. Should be straight-forward.

Will give feedback soon.

brunofavs, 2024-04-04:
Only 1 astra to go :)

[image]

miguelriemoliveira, 2024-04-04:
Great!

miguelriemoliveira, 2024-04-04:
Hi all,

great work.

PLease add the frequency and resolution for each sensor message.

You can use **rostopic hz** to get the frequency.

Also, add the bandwidth **using rostopic bw**

brunofavs, 2024-04-04:
Right now I can :

- Limit the astra's fps 
- Limit L515 color fps (resolution and fps are tied so we can't change them independently)

It's still needed to test : 
- T265 accelerometer and gyroscope fps (already know which parameters to change)
-  Astra resolution (already know which parameters to change)
-  LiDAR rate

But I'm struggling to limit the l515's depth fps.
I searched the sensor's datasheet and there's this table for the color options : 
[image]

But there is no equivalent for depth. Right now the only configuration it is accepting is 30fps. I tried both 6 and 15fps and they all give this error

```
[ WARN] [1712250348.718005134]: Given stream configuration is not supported by the device!  Stream: Depth, Stream Index: 0, Width: 640, Height: 480, FPS: 15, Format: Z16
[ WARN] [1712250348.718072146]: Using default profile instead.
[ INFO] [1712250348.718529698]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
```

I also searched for ways to limit the VLP16 rate and didn't find much.

As a last option we can use a [throttle](http://wiki.ros.org/topic_tools/throttle) node but I would prefer to do it without extra nodes as @manuelgitgomes suggested me.

miguelriemoliveira, 2024-04-04:
> As a last option we can use a [throttle](http://wiki.ros.org/topic_tools/throttle) node but I would prefer to do it without extra nodes as @manuelgitgomes suggested me.

right, that should be a last resort. Try a bit more.

Our target fps is 10 hz for all sensors and maximum available resolution.

Can you set this for the default?

miguelriemoliveira, 2024-04-04:
Another thing missing is the intrinsics calibration for cameras. All should be calibrated using 

https://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration

but this should be done carefully, so lets talk about this next week.

brunofavs, 2024-04-05:
# Updates : 

#### Note

All supported configurations are in  comments at the launch file.

## L515

Went from : 

-  1280x720@6fps RGB
-  640x480@30fps Depth

### to 

- 1920x1080@6fps RGB
- 1024x768@30fps Depth

I exhausted almost all possibilities for the depth, nothing works other than 30fps.

---
## Astras

The Astras can be used in lower resolutions if need be at any framerate, tested already

---
## T265

The accelerometer and gyroscope are fixed at 62.5hz and 200hz respectively. These cannot be changed but they won't impact the bandwidth much.

---
## LiDAR VLP16

Tried messing with `rpm`'s and other launch parameters, nothing changes either the bitrate or the publish rate. It's fixed at 10hz and 2MB/s. This configuration is already good for our needs I think.

# To do now : 

Launch file for recording a bagfile. Will run a couple of tests to determine the current data per unit of time of a recording.

brunofavs, 2024-04-05:
Bagfiles are currently taking roughly 0.5GB/min. I have plenty space available on my pc, I don't think that won't a problem.

```
path:        tmp_2024-04-05-15-48-17.bag
version:     2.0
duration:    14.2s
start:       Apr 05 2024 15:48:17.24 (1712328497.24)
end:         Apr 05 2024 15:48:31.42 (1712328511.42)
size:        124.1 MB
messages:    7058
compression: none [148/148 chunks]
types:       nav_msgs/Odometry           [cd5e73d190d741a2f92e81eda573aca7]
             sensor_msgs/CameraInfo      [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/CompressedImage [8f7a12909da2c9d3332d540a0977563f]
             sensor_msgs/PointCloud2     [1158d486dd51d683ce2f1be655c3c181]
             tf2_msgs/TFMessage          [94810edda583a504dfda3829e70d7eec]
topics:      /astra0/camera_info                              71 msgs    : sensor_msgs/CameraInfo     
             /astra0/image_raw/compressed                     71 msgs    : sensor_msgs/CompressedImage
             /astra1/camera_info                              71 msgs    : sensor_msgs/CameraInfo     
             /astra1/image_raw/compressed                     71 msgs    : sensor_msgs/CompressedImage
             /hand_camera/color/camera_info                   85 msgs    : sensor_msgs/CameraInfo     
             /hand_camera/color/image_raw/compressed          85 msgs    : sensor_msgs/CompressedImage
             /hand_camera/depth/camera_info                  426 msgs    : sensor_msgs/CameraInfo     
             /hand_camera/depth/image_rect_raw/compressed    426 msgs    : sensor_msgs/CompressedImage
             /tf                                            2810 msgs    : tf2_msgs/TFMessage         
             /tf_static                                        1 msg     : tf2_msgs/TFMessage         
             /velodyne_points                                139 msgs    : sensor_msgs/PointCloud2    
             /vo_camera/odom/sample                         2802 msgs    : nav_msgs/Odometry
```

miguelriemoliveira, 2024-04-05:
Hi @brunofavs ,

> I changed the frame ID of the LiDAR to match the T265 odom frame, it was necessary for both to work simultaneously. I think it's a bit confusing but it is what it is.

This does not make sense. We can discuss it Monday ...

miguelriemoliveira, 2024-04-05:
> /hand_camera/depth/image_rect_raw/compressed

This topic can also be a problem, for two reasons:

1. The rect is used when an image rectification is already done. That could be a problem, because that means a calibration is already being done.
2. I had a problem compressing depth data back in 2019 for the atlascar2 . I am not sure this problem exists anymore but we should test it.

brunofavs, 2024-04-05:
Hi @miguelriemoliveira 
> This does not make sense. We can discuss it Monday ...

Ok I will leave it for then.

Regarding the topic for the bagfile:

> 2\. I had a problem compressing depth data back in 2019 for the atlascar2 . I am not sure this problem exists anymore 
but we should test it.

We can also record non compressed image data, it will take more space but its easy to switch.

However,

> 1. The rect is used when an image rectification is already done. That could be a problem, because that means a calibration is already being done.

The depth camera isn't providing anything that isn't rectified, so I'm not sure what to do. Do you know if we can get the non rectified image @manuelgitgomes ?

miguelriemoliveira, 2024-04-05:
Hi think you can use the dynamic reconfigure from rqt to change a parameter to disable the image rectification

Just updated my comment above. Please check.

miguelriemoliveira, 2024-04-05:
One more

/velodyne_points      

should be 

sensor_name/points

We should try not to promote brands in science. They use that as the default but we can change ...

brunofavs, 2024-04-24:
Inesc visit completed
Closing issue