lardemua/atom #919: Creating Zau description (INESC 2024)

manuelgitgomes, 2024-04-04:
I want to create the Zau description, or at least a draft of it, in order to be only a matter of changing values on INESC.

@miguelriemoliveira and @Kazadhum, where did you place the latest version of the zau URDF?
I saw the [zau2 repo](https://github.com/Kazadhum/zau2) and the commits seem very old, do you have it anywhere else?

Tagging @brunofavs for visibility.

Kazadhum, 2024-04-04:
Hi @manuelgitgomes! I'm pretty sure the `zau2_real_description` package in that repo is the latest 'version' before I started trying to change the bagfile instead, so I'd say that's your best bet.

miguelriemoliveira, 2024-04-04:
> I want to create the Zau description, or at least a draft of it, in order to be only a matter of changing values on INESC.

This is a very good idea.

manuelgitgomes, 2024-04-05:
Zeu description written. The robot now looks like this:

[image]

It has:

- UR5 with L515 on hand (`hand_camera`)
- Two astra cameras on body (`front_body_camera` and `back_body_camera`)
- A T265 on body (`left_body_camera`)
- A VLP-16 on body (`lidar3d`)

Is this all correct?

miguelriemoliveira, 2024-04-05:
Hi @manuelgitgomes ,

can you please set colors for the sensors, like in atom examples?

From the softbot experiment I think the back camera is difficult to use. 
Can't we have both looking to the side, even if they do not overlap?

A T265 on body (left_body_camera)

Lets call it body_odom_camera?

About naming, we have to standardize.

I propose:

hand_camera
front_camera
back_camera
odom_camera
boby_lidar

or using the atom examples <modality>_<location> logic

rgbd_hand
rgb_front
rgb_back
odom_body
lidar_body

Tell me which one you'd prefer, and also feel free to suggest changes
Also, the frame_ids should also match these names, but lets first agree on naming ...

miguelriemoliveira, 2024-04-05:
Hi @manuelgitgomes ,

great work. 

> Is this all correct?

It seems fine. Is it an UR5 and not an UR10?

Can you post the tf tree?

miguelriemoliveira, 2024-04-08:
Decision was:

rgbd_hand
rgb_body_front
rgb_body_back
vo_body
lidar_body

manuelgitgomes, 2024-04-26:
This is already done

closed 2024-04-26 by manuelgitgomes