lardemua/atom #921: Bug when running collect_data for rihbot

Kazadhum, 2024-04-09:
Hello!

Ran into what I believe to be a small bug maybe caused by some debug line. When running:

`roslaunch rihbot_calibration collect_data.launch output_folder:=/home/diogo/atom_datasets/rihbot/train_diogo overwrite:=true`

I get:

```
Traceback (most recent call last):4   Duration: 0.154124 / 159.440000               
  File "/home/diogo/catkin_ws/devel/lib/atom_calibration/collect_data", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/diogo/catkin_ws/src/atom/atom_calibration/scripts/collect_data", line 152, in <module>
    data_collector = DataCollector(args, server, menu_handler)
  File "/home/diogo/catkin_ws/src/atom/atom_calibration/src/atom_calibration/collect/data_collector.py", line 132, in __init__
    f = open(transforms_graph_file, 'r')
FileNotFoundError: [Errno 2] No such file or directory: '/home/mike/workspaces/catkin_ws/src/calibration/robots/softbot/softbot_calibration/calibration/transforms_graph.json'
/rgb_hand/image_rawe:   2797.838143   Duration: 0.380143 / 159.440000               
Waiting for camera_info message on topic /rgb_hand/camera_info ... 
received!]  Bag Time:   2797.932112   Duration: 0.474112 / 159.440000               
================================================================================REQUIRED process [collect_data-7] has died!
process has died [pid 13860, exit code 1, cmd /home/diogo/catkin_ws/devel/lib/atom_calibration/collect_data -s 0.5 -o /home/diogo/atom_datasets/rihbot/train_diogo -c /home/diogo/catkin_ws/src/atom/atom_examples/rihbot/rihbot_calibration/calibration/config.yml -ssl lambda sensor_name: False --overwrite __name:=collect_data __log:=/home/diogo/.ros/log/4b76d78a-f65e-11ee-9a33-fbecb084e9fb/collect_data-7.log].
log file: /home/diogo/.ros/log/4b76d78a-f65e-11ee-9a33-fbecb084e9fb/collect_data-7*.log
Initiating shutdown!
================================================================================
```
and RViz crashes. Apparently ATOM is searching for some files related to softbot? Probably some debug line isn't commented.

miguelriemoliveira, 2024-04-09:
I am on it ...

miguelriemoliveira, 2024-04-09:
Better to write this:

    roslaunch rihbot_calibration collect_data.launch output_folder:=/home/diogo/atom_datasets/rihbot/train_diogo overwrite:=true

like this

roslaunch rihbot_calibration collect_data.launch output_folder:=$ATOM_DATASETS/rihbot/train_diogo overwrite:=true

That way it works for both of us

miguelriemoliveira, 2024-04-09:
I confirm the error ...

miguelriemoliveira, 2024-04-09:
Got it. Now we have a new mechanism that produces a transforms_graph.json file which must be read by the data collector. Now it should be working.

closed 2024-04-09 by miguelriemoliveira

Kazadhum, 2024-04-09:
Thank you @miguelriemoliveira ! Just got back from lunch and it's working.

commit a596ef4b (2024-04-09): Fix #921