lardemua/atom #923: Zaubot Intrinsics

brunofavs, 2024-04-11:
[intrinsics.zip](https://github.com/lardemua/atom/files/14943598/intrinsics.zip)

We need to solve the issue that the intel realsense cameras publish a `camera_info` topic but doesn't provide a `set_camera_info`service.

The messy solution is to replace the topic directly in the bagfile, but a  online solution would be preferred.

brunofavs, 2024-04-26:
Hey @miguelriemoliveira @Kazadhum,

Me and @manuelgitgomes calibrated the t265's fisheye cameras intrinsics this morning and we achieved some quite satisfying results.

We found this setting (marked by the red arrow) to be very impactful : 
[image]

After calibration, the pattern grid seemed literally straight as a ruler in all 4 corners, whereas previously was completly rounded.
[t265_fisheye1.tar.gz](https://github.com/lardemua/atom/files/15129525/t265_fisheye1.tar.gz)

[t265_fisheye2.tar.gz](https://github.com/lardemua/atom/files/15129526/t265_fisheye2.tar.gz)

brunofavs, 2024-04-26:
We should be. Didn't we settle on trying to invert the parent/child relationship in the vo_camera odom and renaming the tfs with `rosbag_tools` ?

Kazadhum, 2024-04-26:
Ah yes, I'd forgotten about that

miguelriemoliveira, 2024-04-26:
> Post calibration : https://github.com/lardemua/atom/assets/31831043/5298d189-0289-4443-a367-a5221c0455a2

Looks good. Errors are most of the times under 1 pixel.

brunofavs, 2024-04-26:
How did you conclude that the errors were under 1px?

miguelriemoliveira, 2024-04-26:
In the video there's a number always changing, I think that's the average reprojection error for each frame, which is almost always under 1.0

brunofavs, 2024-05-25:
Already done

closed 2024-05-25 by brunofavs