lardemua/atom #929: Check if nig is working well

miguelriemoliveira, 2024-04-18:
@FYI @brunofavs 

Related to #928 

Make some sanity  tests to make sure noisy initial guess (nig) flag  is working.

brunofavs, 2024-04-18:
Did some more work here.

a57f1652e2d46ae98ab6fa66850a0e45182f3b73

There is still a bug somewhere.

I reckon its inside `compareAtomTransforms()`

With `nig 0.1 0.1`
I couldn't get it to fail. It seems to always be correct, which is easy to mislead us to thinking it's correct.

[image]

However for bigger noises, the rotation fails :

`nig 10 10`

[image]

I did some changes to make sure the noise was being computed, and it still fails with a simpler method (rather than the matrix multiplication thing" 

    import tf
    euler_angles_init = tf.transformations.euler_from_quaternion(transform_1['quat'])
    euler_angles_final = tf.transformations.euler_from_quaternion(transform_2['quat'])

    deuler = np.subtract(euler_angles_final,euler_angles_init)
    rotation_error = np.linalg.norm(deuler)

miguelriemoliveira, 2024-04-18:
> However for bigger noises, the rotation fails 

bigger in rotation, translation, or both?

Is it that you are setting 10 radians for angles, which is larger that 360 degrees?

nig 10 10 is unrealistic. Does it work with  nig 1 3.14?

brunofavs, 2024-04-18:
I should've been more clear.

The bigger the noise is, more often the results are wrong. At 10 rad, it's always wrong from what I tested.
What I cannot utterly understand is that for very low numbers it's accurate.

For numbers usually bigger than 1 its inconsistent.

Runs with `nig 1 3.14` : 
[image]

First image both are wrong
2nd only one   is correct
3rd only the other one is correct
4th both are correct.

I should add that the problem we were having leading to only the `multiple` transformations being displayed correctly I solved already ( in the commit referenced in my previous comment)

brunofavs, 2024-04-18:
It is true that 10rad is outrageous and unrealistic, but it still makes me wonder why it is happening.

While it does not seem to happen for more realistic lower values, I cannot assure that it wont happen given a big enough batch experiment. I might ve simply not ran enough runs to experience it.

brunofavs, 2024-04-24:
from [576b626](https://github.com/lardemua/atom/commit/576b6267a7f2d9461a2bd6128c973a4e532ff5db)

> - Bigger noises are still bugged but now gives a atomWarn()

It's not solved but it will be a lower priority. 

The issue will remain open.