lardemua/atom #935: Dataset creation of Zau

manuelgitgomes, 2024-04-30:
Hello @miguelriemoliveira, @Kazadhum and @brunofavs.

In the next few days, I will be creating the datasets from the bagfiles taken from Zau.
To make the work easier for everyone, I aim to take a dataset that has all the possible information for every use case (be it mine, Diogo's or Bruno's).
For this, I want your opinion on what is needed in the dataset:

- All sensor's messages (including the stereo vo cameras' messages)
- VO odom tf - possibly gathered from altering the tf tree topology, to make it a child instead of a parent.

What more?

brunofavs, 2024-04-30:
Great 
Can you upload the picture as a file instead?

Its really low resolution to see properly 

[image]

manuelgitgomes, 2024-05-03:
Sorry, my bad.
I have now changed the `zau_calibration` package.
You can now check the tree here, accompanied with all the transformations to estimate:
https://github.com/lardemua/zau/blob/main/zau_calibration/calibration/summary.pdf

Please correct it if you see anything wrong, so I can start taking datasets from the bagfiles.

miguelriemoliveira, 2024-05-04:
Hi @manuelgitgomes ,

I was looking at it and it looks good.

Kazadhum, 2024-05-04:
Hello @manuelgitgomes! Looks alright by me!

manuelgitgomes, 2024-05-10:
Hello!

After some unforeseen issues (detailed in #944, miguelriemoliveira/rosbag_tools#5, miguelriemoliveira/rosbag_tools#6), we already have the datasets (without the final bagfile, as I do not have that one yet), before dataset playback.
Some caveats:

- On one bagfile, the messages from the L515 suddenly go "back in time" and stay behind 4000 seconds from the rest of the messages, being unable to take a full collection. I believe the L515 messages, nevertheless, stay coordinated with the rest of the messages. Therefore, I think we can disable that condition on ATOM momentarily to record these message. This is not a very "professional" work, but if we need more collections, that can be one of our resources.
- The world frame is not the same between datasets. This follows our supposition. This implies that the bagfiles cannot be easily stitched together.
- The T265 cannot detect the pattern. I believe this is not a problem on the detector,  as it is universal, i.e., does not depend on intrinsic values. One way this could be solved is by manually annotating the RGB labels.

Will now be labeling the datasets, if any question appears, please ask me.
When they are labeled, I will post more details about the datasets!

miguelriemoliveira, 2024-05-10:
Thanks for the work @manuelgitgomes .

manuelgitgomes, 2024-05-17:
Datasets are created and labeled, closing.

closed 2024-05-17 by manuelgitgomes