lardemua/atom #939: Implement Ali's RWHE calibration method

Kazadhum, 2024-05-03:
The idea is to implement the [RWHE-Calib](https://github.com/ihtishamaliktk/RWHE-Calib) method for Hand-Eye calibration in ATOM.

Similarly to OpenCV, they have two different methods:
- Hand-Eye, which solves the $AX=XB$ equation;
- Robot-World/Hand-Eye, which solves the $AX=ZB$ equation.

Am I correct in assuming we only want the second, @miguelriemoliveira? I remember we spoke of this on Monday because of the OpenCV calibration and if I recall correctly I think we landed on just using RWHE.

By the way, ATOM has a script called "convert_from_rwhe_dataset.py". Are these scripts referring to this method?

miguelriemoliveira, 2024-05-03:
> Am I correct in assuming we only want the second, @miguelriemoliveira? I remember we spoke of this on Monday because of the OpenCV calibration and if I recall correctly I think we landed on just using RWHE.

I think the hand eye should solve both variants, eye-in-hand and eye-to-world. In any case you can use Robot-World/Hand-Eye, I think its better. 

I have been fighting with the opencv's method and still could not make it work.

> By the way, ATOM has a script called "convert_from_rwhe_dataset.py". Are these scripts referring to this method?

Yes, we used Ali's method, but in the matlab scritp.

Kazadhum, 2024-05-03:
> I think the hand eye should solve both variants, eye-in-hand and eye-to-world. In any case you can use Robot-World/Hand-Eye, I think its better.

Ok, that sounds good.
 
> I have been fighting with the opencv's method and still could not make it work.

If you want, I could join you in Zoom after lunch to try to help. Have you made any progress or is it how we left it?

> Yes, we used Ali's method, but in the matlab scritp.

Got it! These scripts will help a lot, I think.

miguelriemoliveira, 2024-05-03:
> If you want, I could join you in Zoom after lunch to try to help. Have you made any progress or is it how we left it?

I changed a lot, trying to cleanup the code. Dit not help. 

I will call if I have some time this afternoon.

Kazadhum, 2024-05-08:
Picking this back up...

Kazadhum, 2024-05-09:
Possible progress! The Li method for calibration in the code base is translated to Python (hopefully correctly!). Now I need to get the comparison working so I can test if it's working properly.
I can't say I understand the code exactly, but I think the translation to Python is correct.

miguelriemoliveira, 2024-05-09:
That's good, but running is the real deal. 

Looking forward to see if it runs correctly.

Kazadhum, 2024-05-13:
Hello all! I think I closed this by accident this morning...

Some notes:
 - I just found out the method I was implementing from this repository was not Ali's, but Li's, so I'll work on Ali's next
 - I got Li's method working for the eye-in-hand case, I'll make another script for the eye-to-hand case

Kazadhum, 2024-05-15:
Working on implementing Shah's method, also from that repo...

Kazadhum, 2024-06-12:
Bug fixing on Li's method after testing on the (simulated) `riwmpbot` revealed issues