lardemua/atom #943: Add a script to use OpenCV's calibration for the eye-to-hand case

Kazadhum, 2024-05-05:
The idea is to create a brother-script to `cv_eye_in_hand.py` which performs the same calibration (with the same four methods but applied to the eye-to-hand case.

I'm going to start working on this now.

Tagging @miguelriemoliveira, @manuelgitgomes and @brunofavs for visibility.

Kazadhum, 2024-05-05:
Looking at the eye-in-hand script and thinking about what needs to change to accommodate the change of variant...

Kazadhum, 2024-05-05:
Simply changing the camera to the pattern in this verification doesn't work, as I suspected, since the getChain() gets the tf chain from the base to the pattern (base to world to pattern, which doesn't go through the hand link). What I can do is check that we aren't in an eye-in-hand configuration, meaning I simply check that, in the chain from the camera link to the base link, the hand link **isn't** there.

The chain should be:
 camera -> ... -> world -> ... -> base

If the hand link belongs to the chain, we know the configuration is wrong.

Kazadhum, 2024-05-05:
The second verification, I'm not so sure about:

https://github.com/lardemua/atom/blob/e28f67e46b386043de75117e2ffe527b97156b69/atom_evaluation/scripts/other_calibrations/cv_eye_in_hand.py#L135-L146

Obviously, it doesn't really work in an eye-to-hand configuration. But should there be another verification here in it's place? Maybe I check if the hand-pattern tf is fixed?

Kazadhum, 2024-05-05:
Added an additional check for if the hand link is in the chain between the pattern link and the base link... I wasn't thinking right before when I said it wouldn't work, but I think the additional check I added is good, if redundant (maybe its "mirror" should be included in the eye-in-hand?)

Kazadhum, 2024-05-05:
This check is not working properly... I forgot the pattern link is not in the transformation pool acquired from the collections. But I think the check can still be made easily. I'll fix it soon.

Kazadhum, 2024-05-05:
Fixed this by checking for the pattern's parent link. That way the getChain() function works properly

Kazadhum, 2024-05-05:
> The second verification, I'm not so sure about: [...]

Again, I wasn't thinking right. We can just check if the TF from the hand link to the pattern is fixed. As above, though, we need to use the pattern link's parent for this:

```
# Check the hand to pattern chain is composed only of fixed transforms
# Since the transformation pool from a collection doesn't include the tf from the pattern link's parent to the pattern link, we must work with the parent. However, it might be the case that the hand link is the same as the pattern's parent link. In that case, it is known that the transform is fixed.

if args['hand_link'] != dataset['calibration_configuration']['calibration_patterns'][args['pattern']]['parent_link']:
    
    chain = getChain(from_frame=args['hand_link'],
                    to_frame=dataset['calibration_config']['calibration_patterns'][args['pattern']]['parent_link'],
                    transform_pool=dataset['collections'][selected_collection_key]['transforms'])

    for transform in chain:
        if not dataset['transforms'][transform['key']]['type'] == 'fixed':
            atomError('Chain from hand link ' + Fore.BLUE + args['hand_link'] + Style.RESET_ALL +
                    ' to pattern link ' + Fore.BLUE +
                    dataset['calibration_config']['calibration_patterns'][args['pattern']]['link'] +
                    Style.RESET_ALL + ' contains non fixed transform ' + Fore.RED +
                    transform['key'] + Style.RESET_ALL + '. Cannot calibrate.')
```

Kazadhum, 2024-05-05:
The next step is to look at the transformations calculated and the ones we effectively want, which will change with regards to the eye-in-hand variant.

miguelriemoliveira, 2024-05-05:
> This doesn't make much sense in the eye-to-hand case, since the hand is not supposed to be in the chain between the base and camera. It is, however, supposed to be in the chain between the base and pattern. I'll change this verification to check for that instead.

Right. Makes sense.

> Obviously, it doesn't really work in an eye-to-hand configuration. But should there be another verification here in it's place? Maybe I check if the hand-pattern tf is fixed?

Perhaps check that the the camera to base chain only contains fixed transforms?

Also, perhaps check that the base to pattern contains the hand chain.
And that the hand to pattern chain transforms are all fixed.

Kazadhum, 2024-05-06:
> Right. Makes sense.
>
>
> Perhaps check that the the camera to base chain only contains fixed transforms?
>
> Also, perhaps check that the base to pattern contains the hand chain. And that the hand to [...]

Implemented!

Now, the OpenCV calibration returns the TFs:
- $^{b}T_p$ (base to pattern);
- $^{h}T_c$ (hand link to camera)

From these TFs, we want to get the atomic tf from the place where the camera is fixed to the camera link.

miguelriemoliveira, 2024-05-06:
> Now, the OpenCV calibration returns the TFs:

this is using the cvcalibrateHandEye? It only returns one transformation, in the case of the eye-in-hand the **hand to camera**, in the case of the eye-to-hand the **camera to base**.

Kazadhum, 2024-05-06:
You're right! The other method returned the other two, I was mistaken

Kazadhum, 2024-05-06:
Running:

`rosrun atom_evaluation cv_eye_to_hand.py -c rgb_world -p pattern_1 -json $ATOM_DATASETS/riwbot/train/dataset.json -ctgt -bl base_link -hl flange`

returns:

```
Partial detection removed: label from collection 001 and pattern pattern_1, sensor rgb_world
Deleted collections: ['001']: at least one detection by a camera should be present.
After filtering, will use 9 collections: ['000', '002', '003', '004', '005', '006', '007', '008', '009']
Selected collection key is 000
Ground Truth b_T_c=
[[-0.876 -0.196  0.44   0.286]
 [ 0.152  0.754  0.639 -0.153]
 [-0.457  0.627 -0.631 -0.355]
 [ 0.     0.     0.     1.   ]]
estimated b_T_c=
[[-0.876 -0.196  0.44   0.286]
 [ 0.152  0.754  0.639 -0.153]
 [-0.457  0.627 -0.631 -0.355]
 [ 0.     0.     0.     1.   ]]
Etrans = 0.0 (m)
Erot = 0.0 (deg)
+--------------------------------------+-------------+---------+----------+-------------+------------+
|              Transform               | Description | Et0 [m] |  Et [m]  | Rrot0 [rad] | Erot [rad] |
+--------------------------------------+-------------+---------+----------+-------------+------------+
| tripod_center_support-rgb_world_link |  rgb_world  |   0.0   | 0.175453 |     0.0     |  1.455959  |
+--------------------------------------+-------------+---------+----------+-------------+------------+
Saved json output file to /home/diogo/atom_datasets/riwbot/train/hand_eye_tsai_rgb_world.json.
```

The good news is the calibration is working correctly and returning the correct tfs, since the comparison between the estimated and GT $^{b}T_c$ are the same. There probably is an issue in the calculation of the tf from the tripod_center_support to the world_link...

Kazadhum, 2024-05-07:
Hi @miguelriemoliveira! So I've been trying to figure this out since this morning... Still didn't get anywhere, but I may have detected an error in the eye-in-hand case.

So one of my doubts was: "If the `b_T_c` transform is correct, then why is the `cp_T_cc` not correct?"

Turns out that maybe `b_T_c` is also not correct. From `cv_eye_in_hand.py`:

https://github.com/lardemua/atom/blob/144c6f0fba5ddaec81668ed79bb3e2a157a84e73/atom_evaluation/scripts/other_calibrations/cv_eye_in_hand.py#L327-L335

Shouldn't we get the ground truth from `dataset_ground_truth` instead of `dataset`?

When I change this the results aren't good, so I wanted to check first.

Kazadhum, 2024-05-07:
Maybe we should return to `cv_eye_in_hand.py` first and then apply whatever fix we find to `cv_eye_to_hand.py`

miguelriemoliveira, 2024-05-07:
> Shouldn't we get the ground truth from dataset_ground_truth instead of dataset?

yes, you are right. But since we did not change the dataset it should be the same, no?

> o when we compare later we're just comparing a transformation against itself, 

ah, if we do this before then we **must** use the ground truth dataset.

> Maybe we should return to cv_eye_in_hand.py first and then apply whatever fix we find to cv_eye_to_hand.py

I agree.

Kazadhum, 2024-05-08:
Investigating now...

Kazadhum, 2024-05-08:
After changing the dataset used for comparison to `dataset_ground_truth`, we get the following results:

```
Ground Truth b_T_c=
[[ 0.922  0.12  -0.369  0.365]
 [ 0.388 -0.285  0.877 -1.444]
 [ 0.    -0.951 -0.309  0.883]
 [ 0.     0.     0.     1.   ]]
estimated b_T_c=
[[-0.876 -0.196  0.44   0.286]
 [ 0.152  0.754  0.639 -0.153]
 [-0.457  0.627 -0.631 -0.355]
 [ 0.     0.     0.     1.   ]]
Etrans = 871.935 (mm)
Erot = 83.585 (deg)
+--------------------------------------+-------------+---------+----------+-------------+------------+
|              Transform               | Description | Et0 [m] |  Et [m]  | Rrot0 [rad] | Erot [rad] |
+--------------------------------------+-------------+---------+----------+-------------+------------+
| tripod_center_support-rgb_world_link |  rgb_world  |   0.0   | 0.175453 |     0.0     |  1.455959  |
+--------------------------------------+-------------+---------+----------+-------------+------------+
Saved json output file to /home/diogo/atom_datasets/riwbot/train/hand_eye_tsai_rgb_world.json.
```

So it seems that the `b_T_c` is wrong and calibrateHandEye is not returning the TF we thought it was.

Kazadhum, 2024-05-08:
Hello @miguelriemoliveira! I (apparently) found out what's wrong!

So, as I commented on #912, nothing was wrong with that, besides a typo in the translation error units. Eye-in-hand is working okay.

About eye-to-hand, now. I had to invert one of the tfs used as input for `cv.calibrateHandEye()`. By inverting `b_T_h` and using `h_T_b` as an input instead, I get:

```
Ground Truth b_T_c=
[[ 0.922  0.12  -0.369  0.365]
 [ 0.388 -0.285  0.877 -1.444]
 [ 0.    -0.951 -0.309  0.883]
 [ 0.     0.     0.     1.   ]]
estimated b_T_c=
[[ 0.922  0.121 -0.369  0.363]
 [ 0.388 -0.288  0.876 -1.439]
 [ 0.    -0.95  -0.312  0.883]
 [ 0.     0.     0.     1.   ]]
Etrans = 1.295 (mm)
Erot = 0.071 (deg)
+--------------------------------------+-------------+---------+----------+-------------+------------+
|              Transform               | Description | Et0 [m] |  Et [m]  | Rrot0 [rad] | Erot [rad] |
+--------------------------------------+-------------+---------+----------+-------------+------------+
| tripod_center_support-rgb_world_link |  rgb_world  |   0.0   | 0.002038 |     0.0     |  0.001232  |
+--------------------------------------+-------------+---------+----------+-------------+------------+
Saved json output file to /home/diogo/atom_datasets/riwbot/train/hand_eye_tsai_rgb_world.json.
```
It is now in compliance with the documentation:

[image]

So I just misread the documentation. I will now rename the variables and check if the comments make sense so the code is cleaner and I'll push it.

Kazadhum, 2024-05-08:
Just finished refactoring the code so it's clear now that it uses `h_T_b` and not `b_T_h`. I believe this calibration method is operational now.

Tagging @miguelriemoliveira and @manuelgitgomes for visibility. Once you have confirmed this is working, this issue can be closed, I think.

miguelriemoliveira, 2024-05-08:
> Tagging @miguelriemoliveira and @manuelgitgomes for visibility. Once you have confirmed this is working, this issue can be closed, I think.

Really happy to ear this.

Congrats!

Kazadhum, 2024-05-08:
Thank you @miguelriemoliveira! I think I'll close this one and try to implement Ali's method next. Afterwards I'll move on to Lidar-camera/Lidar-Lidar calibrations, I think.

closed 2024-05-08 by Kazadhum