lardemua/atom #953: Change ATOM's RViz view from XYOrbit to Orbit

manuelgitgomes, 2024-05-10:
ATOM's RViz view is XYOrbit.
This is the the blue ellipsoid that appears when click on the any button of our mouse.
Example, in the middle of the image:
[image]

This changes the view as we know, but it has a disadvantage, as it is constricted to the ground plane.
There is another view option that is virtually the same, with the ground plane restriction: Orbit
This is represented by a yellow ellipsoid:
[image]

This allows for a much versatile movement inside RViz (although it takes a bit to get used to).
As an example, we can center the view in the middle of the pattern, being easier to label it.

https://github.com/lardemua/atom/assets/58152228/23c79ecd-327c-499d-8bf4-94b6bf9448a4

This is a very simple change in terms of code (a single line), and does not break anything that came beforehand (older version will simply not have this feature automatically added).
Therefore this issue serves more as a discussion if this is worth implementing.

What do you think @miguelriemoliveira, @Kazadhum, @brunofavs?

Kazadhum, 2024-05-10:
Hi! I'd say that if this way is more versatile and facilitates work, then it's worth getting used to, right?

miguelriemoliveira, 2024-05-10:
I have no problem with changing, but I have question: In RViz we can always change the view. 
I are suggesting we change the default view, is that it?