lardemua/atom #954: Implement Lidar to camera Dahll2017

miguelriemoliveira, 2024-05-12:
Paper is LiDAR-Camera Calibration using 3D-3D Point correspondences
https://arxiv.org/pdf/1705.09785

Code is https://github.com/ankitdhall/lidar_camera_calibration/tree/master

miguelriemoliveira, 2024-05-17:
So now we have a tool for annotating lidar developed under #958 and the **rlbot**  lidar data has been annotated.

miguelriemoliveira, 2024-05-17:
Now showing the camera corners (spheres) and the ones given by lidar annotation (cylinders)

[image]

miguelriemoliveira, 2024-05-17:
Calibration using Dahll icp_based method

[image]

miguelriemoliveira, 2024-05-19:
Implemented but I think it works very poorly... @Kazadhum  perhaps we can talk about this ?

Kazadhum, 2024-05-19:
Hi @miguelriemoliveira! Sure, tomorrow at 10am?

miguelriemoliveira, 2024-05-19:
Hi,

Perhaps later in the morning. I will call you ok?

miguelriemoliveira, 2024-05-20:
I was not able to make the Kabsh algorithm work. Tried several implementations. 

For now only the icp_based method works.

miguelriemoliveira, 2024-05-20:
Not working on this anymore.

closed 2024-05-20 by miguelriemoliveira