lardemua/atom #959: Depth labels only save a single border point

manuelgitgomes, 2024-05-15:
Hello @miguelriemoliveira,

Currently, the depth labeler seems to work as intended in data collection:

[image]

However, when the same collection is saved, most of the border points labeled seem to disappear, with the exception of one:
[image]

This happens in all collections.

manuelgitgomes, 2024-05-15:
The solution was way simpler than expected.
The commit diff is self-explanatory really.
You can see it in 3e06363fa4d050ea6f04b203a314a578eef07174.
Basically the for loop that should append the `idx_limit_point`, was appending the fixed `idx`.

closed 2024-05-15 by manuelgitgomes

miguelriemoliveira, 2024-05-15:
Great that you solved it ...