lardemua/atom #963: Using CTRL C to stop dataset playback wiped clean the json.

brunofavs, 2024-05-23:
I used this command to take a few prints of the dataset.

`rosrun atom_calibration dataset_playback -json $ATOM_DATASETS/zau/inesc_day2_5_full/dataset_corrected.json`

Used Ctrl C to exit, it exited cleanly without any error, the json was there but empty.

I'm not sure how to replicate this bug but this behavior though.  If I wasn't using the overwrite flag it shouldn't have done this.
I used this sort of command numerous times before, I never encountered such behavior before.

Due to having the NAS daemon open, it auto overwrote the empty file onto the server.  I asked @manuelgitgomes to do a backup of this file before opening the Synology daemon again and I'm asking whoever sees this issue to do the same. I turned off 2 way synchronization already.

miguelriemoliveira, 2024-05-23:
Sounds strange. Actually I always use the overwrite flag to true, so there might be a bug when its not activated.

miguelriemoliveira, 2024-05-23:
I don't have time to work on this right now ...

brunofavs, 2024-05-23:
The unfortunate thing is that I don't have a reliable way to recreate the issue.
I don't have time for it either, it's okay.

Atleast it's registered in case anyone bumps into it in the future as well.