lardemua/atom #967: Develop Odometry inspector

miguelriemoliveira, 2024-07-11:
Related to #935.

The idea is to create an odometry inspector to allow for the user to manually provide an estimate of the location of the vehicle, thus correction flawed odometry data.

This is useful for fixing the dataset from inesc. 

FYI @Kazadhum .

miguelriemoliveira, 2024-07-11:
First step is done. To show the reference point cloud, so its easy to visually align the robot.

[image]

[image]

miguelriemoliveira, 2024-07-11:
Next step should  be to create an interactive marker to move the robot around and fix the odometry data.

miguelriemoliveira, 2024-07-12:
Hi @Kazadhum and @manuelgitgomes ,

I think I have the odometry inspector up and running.

See this:

https://youtu.be/pIIyoUb_iGE

and give me some feedback.

I will call after lunch.

manuelgitgomes, 2024-07-12:
Looks great, good job!

I have one question that was not clear to be from the video.
What is the default behavior when changing collections?
Does it go to the original odom tf?
Does it change to the same on as the recently changed?
What happens here?

miguelriemoliveira, 2024-07-13:
This is operational. Created a complete corrected dataset for zau in #968.

One nice bonus would be a menu that would run an icp calibration. Will work on it if I have time later ...

miguelriemoliveira, 2024-07-13:
For now closing this one.

closed 2024-07-13 by miguelriemoliveira