lardemua/atom #968: Zau Calibration

manuelgitgomes, 2024-07-12:
Hello @miguelriemoliveira and @Kazadhum 

As you know, I have unsuccessfully tried to calibrate Zau before.
This issue serves for us to solve this problem.

In order to solve this, we need to know what was done beforehand.

Firstly, I tried a full system calibration with:

```
rosrun atom_calibration calibrate -json $ATOM_DATASETS/zau/inesc_day2_5_full/dataset_corrected.json -v -uic -csf 'lambda x: int(x) not in []' -ssf 'lambda x: x not in ["vo_body_fisheye1", "vo_body_fisheye2"]'
```

Which was very slow, so I only retrieved the first iteration, which speaks for itself.

```bash
+------------+----------------+---------------------------------+----------------------------+-----------------------+---------------------+                                                                       
| Collection | lidar_body [m] |        rgb_body_left [px]       |    rgb_body_right [px]     |  rgbd_hand_color [px] | rgbd_hand_depth [m] |                                                                       
+------------+----------------+---------------------------------+----------------------------+-----------------------+---------------------+                                                                       
|    000     |     0.1939     |               ---               |           0.2790           |          ---          |        0.6864       |                                                                       
|    001     |     0.1946     |               ---               |           0.2585           |        740.4794       |        0.3272       |                                                                       
|    002     |     0.3040     |             290.2946            |            ---             |       1544.3119       |        1.4707       |                                                                       
|    003     |     0.3038     |             290.6028            |            ---             |        760.4435       |        0.7229       |                                                                       
|    004     |     0.2496     |             346.4650            |            ---             |       1125.9515       |        1.0762       |                                                                       
|    005     |     0.2514     |             348.3987            |            ---             |        457.5135       |        0.3631       |                                                                       
|    007     |     0.3194     |               ---               |          141.6305          |          ---          |        1.6258       |                                                                       
|    008     |     0.3211     |               ---               |          142.0877          |          ---          |        0.9749       |                                                                       
|    009     |     0.3215     |               ---               |          140.8030          |        552.3251       |        0.3778       |                                                                       
|    010     |     0.3386     |               ---               |          201.1773          |        385.5437       |        0.4151       |                                                                       
|    011     |     0.3359     |               ---               |          200.9437          |        818.3611       |        0.8434       |                                                                       
|    012     |     0.4125     |               ---               |          334.8205          |        512.5840       |        0.2317       |                                                                       
|    013     |     1.1362     |        457821141506.8017        |            ---             |       41065.8179      |        0.6442       |                                                                       
|    014     |     0.9936     |               ---               |         14783.0109         |          ---          |         ---         |                                                                       
|    015     |     0.9894     |               ---               |         14844.2871         |       2385.7146       |        1.1079       |                                                                       
|    016     |     2.0224     |               ---               | 590681790428008022016.0000 | 4249926467125768.0000 |        1.8657       |                                                                       
|    017     |     1.9985     |               ---               |    724095704629585.5000    |       3129.8646       |        1.2227       |                                                                       
|    018     |     1.9973     |               ---               |    758499678056853.7500    |       3360.5949       |        1.2279       |                                                                       
|    019     |     2.1069     |               ---               |    777227631612041.0000    |       3489.5101       |        1.1408       |                                                                       
|    020     |     2.1440     | 47906817194096436950597632.0000 |            ---             |     355103825.9580    |        2.8356       |                                                                       
|    021     |     2.1603     |   1148544782966324985856.0000   |            ---             |     14331041.9931     |        1.9194       |                                                                       
|    022     |     2.1723     |   1148544782966324985856.0000   |            ---             |     14318005.4360     |        1.9197       |
|    024     |     2.1337     |               ---               |        1191711.6395        |          ---          |        0.3958       |
|    025     |     2.1174     |               ---               |        1116122.1382        |        885.2455       |        1.5117       |
|    026     |     2.6738     |               ---               |       11687309.1259        |      386324.8061      |        1.8028       |
|    027     |     2.6722     |               ---               |       10984409.7375        |      491618.1919      |        1.8181       |
|    028     |     2.6772     |               ---               |       11150094.8619        |  63638306458559.6250  |        1.9609       |
|    029     |     2.7334     |             886.0851            |            ---             |          ---          |        2.4232       |
|    030     |     2.7336     |             889.6913            |            ---             |       88040.1836      |        1.9992       |
|    031     |     2.7362     |             892.0146            |            ---             |          ---          |        2.0286       |
|    032     |     2.7642     |             892.0168            |            ---             |       37231.9843      |        2.0115       |
|    033     |     3.0139     |             716.5487            |            ---             |       4658.1998       |        2.3001       |
|    034     |     3.0400     |             735.9727            |            ---             |       67295.5261      |        1.5374       |
|    035     |     3.0144     |             735.2294            |            ---             |          ---          |        1.2621       |
|    036     |     3.0396     |             735.2112            |            ---             |       21235.0739      |        2.0732       |                                                                       
|    037     |     2.7757     |               ---               |         1207.6261          |          ---          |        0.4974       |                                                                       
|    038     |     2.7790     |               ---               |         1160.5614          |  115674160710200.3438 |        1.9120       |                                                                       
|    039     |     3.2655     |            1889.2326            |            ---             |        816.1449       |        2.9535       |                                                                       
[... cut ...]
|    053     |        ---         |       15.0536       |
|    054     |        ---         |       15.0672       |
|    055     |        ---         |       16.1876       |
|    056     |        ---         |       15.0845       |
|    057     |        ---         |       15.0779       |
|    058     |      14.4713       |         ---         |
|    059     |      14.6249       |         ---         |
|  Averages  |      82.5107       |       78.1748       |
+------------+--------------------+---------------------+
```

miguelriemoliveira, 2024-07-13:
Thanks @manuelgitgomes for the detailed description. It should allow me to replicate the experiments with the odometry corrected and evaluate the differences.

One question: are you using @brunofavs approach of calibrating the odometry transformation?

manuelgitgomes, 2024-07-13:
Hello!

These results are with Bruno's approach!

One question: are you using @brunofavs<https://github.com/brunofavs> approach of calibrating the odometry transformation or these results are all without calibrating the odometry?

miguelriemoliveira, 2024-07-13:
Thanks @manuelgitgomes .

I just finished correcting the odometry in the zau dataset ... 59 collections ... uff.

Now I will try to calibrate and post results asap.

miguelriemoliveira, 2024-07-13:
> I also tried with rgb_body_right:

Before Averages  |  273255929619060.7188 
Now Averages  |        0.3023       

> Then, I tried only using a small number of collections:

Before Averages  |        0.3358       
Now Averages  |        0.3302       |

> I suspect this small group of collections were all with consistent odometry as it only jumps from time to time.
> Which after 850 iterations gave me the results:

Before Averages  |      82.5107       |       78.1748   
Now Averages  |       0.3663       |        0.3157       

Later on I will try the full calibration, but for now it looks good.

miguelriemoliveira, 2024-07-13:
Hi @manuelgitgomes ,

> These results are with Bruno's approach!

But I was looking into the calibration/config.yaml file and in the additional tfs the odometry itransform s not included.

# ATOM will calibrate the extrinsic parameters of your links.
# In this section you should discriminate the additional transformations that will be part of the calibrations.
additional_tfs:
  # Each key will define a transformations and its name, which will be use throughout the calibration.
  # Each additional transformations definition must have the following properties:
  #
  #       parent_link:
  #           The parent link of the transformation (i.e. link) to be calibrated.
  #
  #       child_link:
  #           This is the transformation (i.e. link) that we be optimized.
  #
  # EXAMPLE:
  base_link_mb_to_base_link:
   parent_link: "base_link_mb"
   child_link: "base_link"x

[Here also ](https://github.com/lardemua/zau/blob/main/zau_calibration/calibration/config.yml)

That means we are not calibrating the odometry, right? 
Perhaps you ran before with a different config.yml file?

miguelriemoliveira, 2024-07-13:
Hi again @manuelgitgomes ,

So I when to look at the dataset_corrected.json and in fact you have the calibration of the **world to base_footprint** transform.

[image]

It seems the config.yml is not up to date with the datasets. Did you forget to push something?

miguelriemoliveira, 2024-07-13:
Results for two body cameras + lidar (taking some collections away)

    rosrun atom_calibration calibrate -json $ATOM_DATASETS/zau/inesc_day2_5_full/dataset_corrected.json -v -uic -csf 'lambda x: int(x) not in [11,12, 30,33,51, 52]' -ssf 'lambda x: x not in ["vo_body_fisheye1", "vo_body_fisheye2", "rgbd_hand_color", "rgbd_hand_depth"]'

```
+------------+----------------+--------------------+---------------------+
| Collection | lidar_body [m] | rgb_body_left [px] | rgb_body_right [px] |
+------------+----------------+--------------------+---------------------+
|  Averages  |     0.0099     |       1.5042       |        2.1462       |
```

which is quite good.

brunofavs, 2024-07-13:
`Very good news. It means the odometry calibration is doing a lot of the work!`

Indeed, these results are great :)

manuelgitgomes, 2024-07-15:
Hello!

> That means we are not calibrating the odometry, right?
> Perhaps you ran before with a different config.yml file?

I am sorry, just checked I have uncommitted files.
That should now be solved.

> which is quite good.

These results are great!
This is going in the right direction.

miguelriemoliveira, 2024-07-15:
> I am sorry, just checked I have uncommitted files.
> That should now be solved.

Humm ... I still do not have the config.yml listing the odometry transformation ...

manuelgitgomes, 2024-07-15:
@miguelriemoliveira Well, then I could have added the additional_tf manually to the dataset :sweat_smile: 
If you want me to reconfigure zau, I can do it

miguelriemoliveira, 2024-07-15:
> Well, then I could have added the additional_tf manually to the dataset

You do this to avoid having to create a new dataset right?

> If you want me to reconfigure zau, I can do it

No need. I can do it. Thanks.

manuelgitgomes, 2024-07-15:
> You do this to avoid having to create a new dataset right?

Exactly.

Kazadhum, 2024-07-15:
Just finished reading the issue and these results are a big improvement. Sounds like good progress!

miguelriemoliveira, 2024-07-19:
Closing this issue. Will open a new one for the depth camera related issues (#975).

closed 2024-07-19 by miguelriemoliveira