lardemua/atom #970: Is the camera in the correct place in the zau inesc datasets?

miguelriemoliveira, 2024-07-13:
FYI @manuelgitgomes and @Kazadhum ,

I don't think it is ... I mean when correcting the odometry I had strange projections like in these examples:

[image]

from this one it seems that the robot arm is not correctly aligned 

[image]

This one is also strange

[image]

Here's the positioning of the camera

[image]

miguelriemoliveira, 2024-07-13:
I tried to calibrate with hand camera rgb + another rgb and the calibration does not converge.
I think this is the problem. Initial guess for the depth is very wrong ...

manuelgitgomes, 2024-07-15:
Hello @miguelriemoliveira!

I am fairly certain the hand camera is rotated 180º in roll.
That should be fairly easy to rotate hardcoded in the dataset.
But, if you have the will to work on #971, that would be nice as well.

miguelriemoliveira, 2024-07-15:
> I am fairly certain the hand camera is rotated 180º in roll.
> That should be fairly easy to rotate hardcoded in the dataset.
> But, if you have the will to work on #971, that would be nice as well.

Thanks @manuelgitgomes . I think #971 is better because usually my 3 month long searches for the correct transformation always start like this: I am fairly certain that we need to rotate this x degrees ... :)

Kazadhum, 2024-07-15:
Ah, it looks like I might have made a mistake when inserting the camera in the URDF when we were over at INESC... :sweat_smile:

miguelriemoliveira, 2024-07-15:
> Ah, it looks like I might have made a mistake when inserting the camera in the URDF when we were over at INESC... 😅

Yeah, that happens to everyone. That's why we need #971

miguelriemoliveira, 2024-07-17:
Correcting using transform_corrector.

[image]

[image]

It's not perfect, but much better.

miguelriemoliveira, 2024-07-17:
It was mostly about rotating the camera 180 degrees.

closed 2024-07-17 by miguelriemoliveira