lardemua/atom #971: Generalize odometry inspector

miguelriemoliveira, 2024-07-13:
Because of #970 we now seem to have problems in the initial transforms for another sensor.

That means we have to generalize the odometry inspector to run be able to cope with any transformation.

Ultimately we will remove the set initial estimate working from the bagfile, and create a dataset_playbag 2.0 that will allow us to correct annotations and manually set any transformations we'd like.

Kazadhum, 2024-07-17:
Hi @miguelriemoliveira! 

Indeed, I wrote a check for if a given TF was fixed! It's not a separate function, but its here:

https://github.com/lardemua/atom/blob/3d8558fde888b8c079b87c845639e9fe2503404c/atom_evaluation/scripts/other_calibrations/copy_tfs_from_dataset.py#L71-L89

Keep in mind this comes in the context of the `copy_tfs_from_dataset.py` script, where the child links and corresponding parent links are given as input arguments. (which are then reorganized into `(parent, child)` tuples for organization).

I can reorganize it into a function quickly if needed!

miguelriemoliveira, 2024-07-17:
This is were I have done it before

https://github.com/lardemua/atom/blob/b1f0a423e92cde78afb850863ee2ebbd9b284a1b/atom_calibration/src/atom_calibration/calibration/visualization.py#L496

miguelriemoliveira, 2024-07-17:
I think all is done for now. I have use the dataset['transforms'] that exist in newer datasets. 

Perhaps we will need to create a dataset migrate tool ... opened #973 for this.

miguelriemoliveira, 2024-07-17:
Missing the creation of a 

transform_corrector.launch ... should be done in the configure_calibration_package.
Opened #974 for this.