lardemua/atom #972: Should getMixedDataset() copy the pattern's TFs from the train dataset?

Kazadhum, 2024-07-15:
It makes sense to me that for a fixed pattern, we use the calibrated pattern TFs for evaluation. What do you think @miguelriemoliveira and @manuelgitgomes?

miguelriemoliveira, 2024-07-15:
Hi @Kazadhum ,

I think this is not always true. Consider the softbot, for example. The pattern is fixed, but from one dataset to another it may be changed.

So I would not assume that the pattern transformations can always be copied for fixed patterns. 

Perhaps we can use a flag to activate this copy?

Kazadhum, 2024-07-16:
Hi @miguelriemoliveira!

You're right. We can't and shouldn't make that assumption. The flag is already implemented I think, as that's exactly what `-cpt` does.

miguelriemoliveira, 2024-07-16:
Close issue?

closed 2024-07-16 by Kazadhum