lardemua/atom #978: Error with OpenCV calibration with specific pattern and method

Kazadhum, 2024-07-19:
I'm running into an error when running the OpenCV calibration for RIWMPBOT (simulated), specifically for the upper arm pattern and the `andreff` method.

When running:

```
rosrun atom_evaluation cv_eye_to_hand.py \
-json $ATOM_DATASETS/riwmpbot/train/dataset.json \
-c rgb_world -p upper_arm_pattern \
-hl upper_arm_link -bl base_link \
-csf 'lambda x: True' \
-ctgt -uic -mn andreff \
-sfr -sfrn /tmp/comparison_to_gt.csv
```

I get:

```
Traceback (most recent call last):
  File "/home/diogo/catkin_ws/src/atom/atom_evaluation/scripts/other_calibrations/cv_eye_to_hand.py", line 518, in <module>
    main()
  File "/home/diogo/catkin_ws/src/atom/atom_evaluation/scripts/other_calibrations/cv_eye_to_hand.py", line 398, in main
    b_T_c_R, b_T_c_t = cv2.calibrateHandEye(h_T_b_Rs,
cv2.error: OpenCV(4.6.0) /io/opencv/modules/calib3d/src/calibration_handeye.cpp:524: error: (-7:Iterations do not converge) Rotation normalization issue: determinant(R) is null in function 'normalizeRotation'
```

Kazadhum, 2024-07-19:
I found this entry in stack overflow: https://stackoverflow.com/questions/72335499/wrong-result-in-cv2-calibratehandeye

The user seems to have encountered the same problem, also with the Andreff method

Kazadhum, 2024-07-19:
Maybe this is not an error and is instead just a limitation of the Andreff method

miguelriemoliveira, 2024-07-20:
I think you are right. Just add a note in the table saying "did not converge".

closed 2024-07-23 by Kazadhum