lardemua/atom #982: Bug with JointTypes during Zau calibration

Kazadhum, 2024-09-10:
This came while testing the Zau calibration process.

Running:

```
clear && \
export CSF="lambda x: int(x) not in []" \
&& rosrun atom_calibration calibrate \
-json $ATOM_DATASETS/zau_real/inesc_day2_5_full/dataset_corrected_with_odometry_and_depth.json -v -uic \
-csf "$CSF" \
-jsf "lambda x: False" -jpsf "lambda x: x in ['origin_yaw']" -ssf "lambda x: x not in ['vo_body_fisheye1','vo_body_fisheye2']"
```

results in:

```
Traceback (most recent call last):
  File "/home/diogo/catkin_ws/devel/lib/atom_calibration/calibrate", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/diogo/catkin_ws/src/atom/atom_calibration/scripts/calibrate", line 874, in <module>
    main()
  File "/home/diogo/catkin_ws/src/atom/atom_calibration/scripts/calibrate", line 264, in main
    dataset = filterJointsFromDataset(dataset, args)
  File "/home/diogo/catkin_ws/src/atom/atom_core/src/atom_core/dataset_io.py", line 751, in filterJointsFromDataset
    for joint_key in dataset['calibration_config']['joints']:
TypeError: 'NoneType' object is not iterable
```

Working on the fix...

Tagging for visibilty @miguelriemoliveira @brunofavs

Kazadhum, 2024-09-10:
I found the error.

In `dataset_io.py`:

https://github.com/lardemua/atom/blob/498b54f02fd3d771e3704d7be11d5f8a72d89ffc/atom_core/src/atom_core/dataset_io.py#L741-L762

If we pass -jsf as argument, even if it's `"lambda x: False"` then filtering doesn't work if there are no joints defined in the dataset. Running it without the -jsf argument works. This probably happens because of different dataset versions containing different things. Regardless, I think the if statements can be changed to accommodate for older datasets without the "joints" field.

The same thing happens with the -jpsf argument.

closed 2024-09-10 by Kazadhum

miguelriemoliveira, 2024-09-10:
hum ... strange ... why don't I have this problem?

Kazadhum, 2024-09-10:
I fixed it by changing the `filterJointsFromDataset()` function to first check if there are joints defined in the dataset before attempting to iterate through them.

https://github.com/lardemua/atom/blob/92153879571c34dfc8bf40526228aac27b87301e/atom_core/src/atom_core/dataset_io.py#L742-L764

I made the same change to `filterJointParametersFromDataset()`:

https://github.com/lardemua/atom/blob/92153879571c34dfc8bf40526228aac27b87301e/atom_core/src/atom_core/dataset_io.py#L713-L739

Kazadhum, 2024-09-10:
> hum ... strange ... why don't I have this problem?

Not sure... The error makes sense to me though