lardemua/atom #983: List of preliminary assessments before doing batches for Odometry paper

brunofavs, 2024-09-10:
As we're coming back from the summer, in order to freshen up the point where we left things off with ZAU, I wrote a list of preliminary tests we should do before committing to doing long experiment batches.

Considering all the variables we have at play :
## Variables

### Datasets
- [ ] Dataset without corrected odometry
- [ ] Dataset with corrected odometry
- [ ] Dataset with corrected odometry and depth

### Which sensors to calibrate:

Steps:

- [ ] 1 RGB
- [ ] 2 RGB
- [ ] 3 RGB + Depth ( the 3rd RGB comes from the rgbd camera)
- [ ] 3 RGB + Depth + LiDAR

Fisheyes are out because we don't have a method to manually label them.

### Calibrating odometry

 - [ ]  With and without my MsC. thesis methodology

Tagging @manuelgitgomes @miguelriemoliveira @Kazadhum for visibility.

Kazadhum, 2024-09-10:
We're also expecting to have to correct the hand RGB camera pose in the initial guess

miguelriemoliveira, 2024-09-10:
Makes sense.

brunofavs, 2024-09-11:
Hey, last night I ran some of the tests mentioned in the checklist.

These tests were run with the dataset @miguelriemoliveira sent me and @Kazadhum yesterday by email. The one with corrected odometry and depth orientation.

# Comands for each test

#### Dataset corrected

1 RGB

```bash
export CSF="lambda x: int(x) not in []" \
export DATASET=$ATOM_DATASETS/zau/inesc_day2_5_full/dataset_corrected.json
&& rosrun atom_calibration calibrate \
-json $DATASET -uic -v  \
-csf "$CSF" -ssf "lambda x: x in ['rgb_body_left']" \
 -ftol 1e-4 -xtol 1e-4 -gtol 1e-4
 ```

2 RGB

```bash
export CSF="lambda x: int(x) not in []" \
export DATASET=$ATOM_DATASETS/zau/inesc_day2_5_full/dataset_corrected.json
&& rosrun atom_calibration calibrate \
-json $DATASET -uic -v  \
-csf "$CSF" -ssf "lambda x: x in ['rgb_body_left','rgb_body_right']" \
 -ftol 1e-4 -xtol 1e-4 -gtol 1e-4
 ```

3 RGB + Depth

```bash
export CSF="lambda x: int(x) not in []" \
export DATASET=$ATOM_DATASETS/zau/inesc_day2_5_full/dataset_corrected.json
&& rosrun atom_calibration calibrate \
-json $DATASET -uic -v  \
-csf "$CSF" -ssf "lambda x: x in ['rgb_body_left','rgb_body_right','rgbd_hand_color','rgbd_hand_depth']" \
 -ftol 1e-4 -xtol 1e-4 -gtol 1e-4
 ```
 
3 RGB + Depth + LiDAR

```bash
export CSF="lambda x: int(x) not in []" \
export DATASET=$ATOM_DATASETS/zau/inesc_day2_5_full/dataset_corrected.json
&& rosrun atom_calibration calibrate \
-json $DATASET -uic -v  \
-csf "$CSF" -ssf "lambda x: x in ['rgb_body_left','rgb_body_right','rgbd_hand_color','rgbd_hand_depth','lidar_body']" \
 -ftol 1e-4 -xtol 1e-4 -gtol 1e-4
 ```
 
#### Dataset corrected with 

1 RGB

```bash
export CSF="lambda x: int(x) not in []" \
export DATASET=$ATOM_DATASETS/zau/inesc_day2_5_full/dataset_corrected_with_odometry.json && rosrun atom_calibration calibrate \
-json $DATASET -uic -v  \
-csf "$CSF" -ssf "lambda x: x in ['rgb_body_left']" \
 -ftol 1e-4 -xtol 1e-4 -gtol 1e-4
 ```

2 RGB

```bash
export CSF="lambda x: int(x) not in []" \
export DATASET=$ATOM_DATASETS/zau/inesc_day2_5_full/dataset_corrected_with_odometry.json && rosrun atom_calibration calibrate \
-json $DATASET -uic -v  \
[... 250 lines cut ...]
    parent_link: "world" 

    fixed: true

    pattern_type: "charuco"

    dictionary: "DICT_5X5_100"

    mesh_file: "package://atom_worlds/pattern/models/charuco_800x600_5x5_100/charuco_800x600_5x5_100.dae"

    border_size: { 'x': 0.04, 'y': 0.03 }

    dimension: { "x": 11, "y": 8 }
    size: 0.06
    inner_size: 0.045
	
anchored_sensor:

max_duration_between_msgs: 1000

package_name: "zau_calibration"

version: 3.0

```

brunofavs, 2024-09-11:
# A few notes about the last comment

I did forget to change the `additional_tfs` in the configuration yaml, right now the ones defined don't represent any transformation in the tf_tree. I don't know if it affects something.

I think I'm not doing something right. When the depth is added, the calibration goes bad, however the runs with only RGB sensors results don't differ from using the dataset with and without the corrected odometry. They already start with a really small residual and so they finish with a small residual. I'm not sure why this is the case.

I played back the dataset with corrected odometry and it seemed fine, the odometry was in fact corrected. I should check the dataset without corrected odometry as well.

miguelriemoliveira, 2024-09-11:
> These tests were run with the dataset @miguelriemoliveira sent me and @Kazadhum yesterday by email. The one with corrected odometry and depth orientation.

I don't get it. You say that you used the dataset with odometry and depth corrected, but then present different datasets?

miguelriemoliveira, 2024-09-11:
> I did forget to change the `additional_tfs` in the configuration yaml, right now the ones defined don't represent any transformation in the tf_tree. I don't know if it affects something other than the transformation given by the odometry not being estimated.

Yes, I think you must repeat all the results.

 
> I think I'm not doing something right. When the depth is added, the calibration goes bad, however the runs with only RGB sensors results don't differ from using the dataset with and without the [...]

I see very different values in the two datasets ...

> I played back the dataset with corrected odometry and it seemed fine, the odometry was in fact corrected. I should check the dataset without corrected odometry as well.

We must be careful with the naming. Which dataset are you using? What is the file name?

brunofavs, 2024-09-11:
> You also have a title called **Dataset corrected with** ??? What does this mean?

I meant "Dataset corrected with odometry", I miswrote.

> > I played back the dataset with corrected odometry and it seemed fine, the odometry was in fact corrected. I should check the dataset without corrected odometry as well.
> 
> We must be careful with the naming. Which dataset are you using? What is the file name?

"dataset_corrected.json" and "dataset_corrected_with_odometry.json". I do find the naming here easy to mess up considering "corrected" doesn't mean that the dataset is corrected and thus good to use.

miguelriemoliveira, 2024-09-11:
I think the dataset you should use is only the **dataset_corrected_with_odometry_and_depth.json**.

That's the one that should  give good results.

brunofavs, 2024-09-11:
I'm going to repeat these with after fixing the `additional_tfs` field. Also I have all these datasets :

-  `atom_calibration.json`
-  `dataset.json`
-  `dataset_corrected.json`
-  `dataset_corrected.json.bak`
-  `dataset_corrected_with_odometry.json`
-  `dataset_corrected_with_odometry_and_depth.json`
-  `dataset_corrected_without_odometry.json`

Whats the difference between the `dataset_corrected.json` and `dataset_corrected_without_odometry.json`? I played back both of those and couldn't pinpoint what was different. The following images are from the 2nd collection of each dataset.

`dataset_corrected.json`
[image]

`dataset_corrected_without_odometry.json`
[image]

miguelriemoliveira, 2024-09-11:
> Whats the difference between the dataset_corrected.json and dataset_corrected_without_odometry.json? I played back both of those and couldn't pinpoint what was different. The following images are from the 2nd collection of each dataset.

I think no difference. In any case only the 

dataset_corrected_with_odometry_and_depth.json

is to be used I think.

Later on, @Kazadhum should produce a new dataset called

dataset_corrected_with_odometry_depth_and_hand_rgb.json

which should be the one we use in the paper.

brunofavs, 2024-09-11:
Hey,
Here are the results from the `dataset_corrected_with_odometry_and_depth.json`. Will redo with the `dataset_corrected_with_odometry_and_depth_and_rgb.json` from @Kazadhum now.

The `config.yml` was updated so now all these calibrations are including the estimation of the TF given by the odometry system.

The commands for the calibrations from which the following residual tables were generated are equal to the ones above on this thread except for the dataset used, so I'll spare them to make it easier to read.

# dataset_corrected_with_odometry_and_depth.json

1 RGB

```bash
+------------+--------------------+
| Collection | rgb_body_left [px] |
+------------+--------------------+
|    059     |       0.3547       |
|  Averages  |       0.4098       |
+------------+--------------------+

 ```

2 RGB

```bash
+------------+--------------------+---------------------+
| Collection | rgb_body_left [px] | rgb_body_right [px] |
+------------+--------------------+---------------------+
|    058     |       0.3895       |         ---         |
|    059     |       0.4016       |         ---         |
|  Averages  |       0.9745       |        0.4144       |
+------------+--------------------+---------------------+

 ```

2 RGB + Depth

```bash
+------------+--------------------+---------------------+---------------------+
| Collection | rgb_body_left [px] | rgb_body_right [px] | rgbd_hand_depth [m] |
+------------+--------------------+---------------------+---------------------+
|    058     |      10.2244       |         ---         |        0.0075       |
|    059     |       5.1885       |         ---         |        0.0080       |
|  Averages  |      10.8728       |        5.5046       |        0.0064       |
+------------+--------------------+---------------------+---------------------+

 ```

2 RGB + LiDAR

```bash
+------------+----------------+--------------------+---------------------+
| Collection | lidar_body [m] | rgb_body_left [px] | rgb_body_right [px] |
+------------+----------------+--------------------+---------------------+
|    059     |     0.0068     |       5.1291       |         ---         |
|  Averages  |     0.0175     |       7.0713       |        6.7862       |
+------------+----------------+--------------------+---------------------+
 ```

3 RGB + Depth

```bash
+------------+--------------------+---------------------+----------------------+---------------------+
| Collection | rgb_body_left [px] | rgb_body_right [px] | rgbd_hand_color [px] | rgbd_hand_depth [m] |
+------------+--------------------+---------------------+----------------------+---------------------+
|    058     |      10.7600       |         ---         |        3.9036        |        0.0075       |
|    059     |       5.7394       |         ---         |        1.8524        |        0.0080       |
|  Averages  |      11.6182       |        5.9879       |        4.5981        |        0.0063       |
+------------+--------------------+---------------------+----------------------+---------------------+

 ```

3 RGB + Depth + LiDAR

```bash
+------------+----------------+--------------------+---------------------+----------------------+-------------------- +
| Collection | lidar_body [m] | rgb_body_left [px] | rgb_body_right [px] | rgbd_hand_color [px] | rgbd_hand_depth [m] |
+------------+----------------+--------------------+---------------------+----------------------+-------------------- +
|    059     |     0.0079     |       6.6278       |         ---         |        2.5710        |        0.0085       |
|  Averages  |     0.0136     |      14.2390       |        9.1803       |        5.3333        |        0.0095       |
+------------+----------------+--------------------+---------------------+----------------------+---------------------+

 ```

Every calibration yielded good results as per my understanding. When adding a second modality to the calibration, the residuals seem to get worse.

brunofavs, 2024-09-11:
# Update
## Results from `dataset_corrected_with_odometry_and_depth_and_rgb.json`

- Odometry transformations were estimated;
- Same commands from above with different dataset

Pretty good results overall.

Waiting for your feedback @miguelriemoliveira @Kazadhum 

### 1 RGB

```yaml
+------------+--------------------+
| Collection | rgb_body_left [px] |
+------------+--------------------+
|    000     |        ---         |
|    001     |        ---         |
|    002     |       0.2665       |

|  Averages  |       0.4098       |
+------------+--------------------+
 ```

### 2 RGB

```yaml
+------------+--------------------+---------------------+
| Collection | rgb_body_left [px] | rgb_body_right [px] |
+------------+--------------------+---------------------+
|    000     |        ---         |        0.3259       |
|    001     |        ---         |        0.2630       |
|    002     |       0.9531       |         ---         |

|  Averages  |       0.9745       |        0.4144       |
+------------+--------------------+---------------------+
 ```

### 2 RGB + LiDAR

```yaml
+------------+----------------+--------------------+---------------------+
| Collection | lidar_body [m] | rgb_body_left [px] | rgb_body_right [px] |
+------------+----------------+--------------------+---------------------+
|    000     |     0.0099     |        ---         |        7.2151       |
|    001     |     0.0069     |        ---         |        6.3340       |
|    002     |     0.0076     |       4.7857       |         ---         |

|  Averages  |     0.0175     |       7.0730       |        6.7843       |
+------------+----------------+--------------------+---------------------+
 ```

### 2 RGB + Depth

```yaml
+------------+--------------------+---------------------+---------------------+
| Collection | rgb_body_left [px] | rgb_body_right [px] | rgbd_hand_depth [m] |
+------------+--------------------+---------------------+---------------------+
|    000     |        ---         |        7.9810       |        0.0050       |
|    001     |        ---         |        8.7077       |        0.0061       |
|    002     |       3.1255       |         ---         |        0.0078       |

|  Averages  |      10.3961       |        5.0947       |        0.0063       |
+------------+--------------------+---------------------+---------------------+

 ```

### 3 RGB + Depth

```yaml
+------------+--------------------+---------------------+----------------------+---------------------+
| Collection | rgb_body_left [px] | rgb_body_right [px] | rgbd_hand_color [px] | rgbd_hand_depth [m] |
+------------+--------------------+---------------------+----------------------+---------------------+
|    000     |        ---         |        9.1064       |         ---          |        0.0049       |
|    001     |        ---         |        9.0890       |        3.2440        |        0.0061       |
|    002     |       3.0586       |         ---         |        4.2052        |        0.0078       |

|  Averages  |      10.8815       |        5.4606       |        4.4225        |        0.0063       |
+------------+--------------------+---------------------+----------------------+---------------------+
 ```
 
### 3 RGB + Depth + LiDAR

```yaml
+------------+----------------+--------------------+---------------------+----------------------+---------------------+
| Collection | lidar_body [m] | rgb_body_left [px] | rgb_body_right [px] | rgbd_hand_color [px] | rgbd_hand_depth [m] |
+------------+----------------+--------------------+---------------------+----------------------+---------------------+
|    000     |     0.0096     |        ---         |       17.2694       |         ---          |        0.0062       |
|    001     |     0.0087     |        ---         |       11.6445       |        3.8885        |        0.0067       |
|    002     |     0.0099     |       3.6430       |         ---         |        4.3975        |        0.0082       |

|  Averages  |     0.0136     |      11.6325       |        7.1985       |        5.0736        |        0.0096       |
+------------+----------------+--------------------+---------------------+----------------------+---------------------+
 ```

miguelriemoliveira, 2024-09-12:
About the results with the **dataset_corrected_with_odometry_and_depth.json**, overall they seem fine.

However, I do not understand why when we use the rgb hand camera the results are ok (4 pixels error).

Shouldn't we have very bad values?
Is that not the reason we need the 
**dataset_corrected_with_odometry_and_depth_and_rgb.json**?

About the results with the **dataset_corrected_with_odometry_and_depth_and_rgb.json**,  again overall they seem fine, but the rgb hand gives worse results than with the **dataset_corrected_with_odometry_and_depth.json**.
Very strage ...

Can you run a calibration only with the body left and the hand camera?

miguelriemoliveira, 2024-09-12:
Are there any collections you think should be removed?

brunofavs, 2024-09-12:
I just sent you a email about something else :D Right on time

> Are there any collections you think should be removed?

I will rerun a calibration to see if there are some problematic collections, I don't remember.

> Can you run a calibration only with the body left and the hand camera?

Will do.

> Shouldn't we have very bad values?

That was also my main concern.. I'm not sure why that is..

Kazadhum, 2024-09-12:
> However, I do not understand why when we use the rgb hand camera the results are ok (4 pixels error).
> 
> Shouldn't we have very bad values? Is that not the reason we need the **dataset_corrected_with_odometry_and_depth_and_rgb.json**?

Is it possible that just correcting one of the cameras' pose initial guess facilitates the optimization enough that it can overcome the RGB camera's poor initial guess?

> About the results with the **dataset_corrected_with_odometry_and_depth_and_rgb.json**, again overall they seem fine, but the rgb hand gives worse results than with the **dataset_corrected_with_odometry_and_depth.json**. Very strage ...

Hmm, this does seem strange... I'd say maybe we're seeing some overfitting but I don't think so, since the reprojection error is used for optimization

Kazadhum, 2024-09-12:
> > Are there any collections you think should be removed?
> 
> I will rerun a calibration to see if there are some problematic collections, I don't remember.

I think problematic collections are easier to identify with `dataset_playback`, maybe... I did this two days ago but I didn't find any weird ones, but I might have missed something!

brunofavs, 2024-09-12:
One thing that is also bothering me is that the residuals are starting already way too small. In the `dataset_corrected_with_odometry_and_depth_and_rgb` in the calibration with 2 RGB for example, the residuals start at 0.9 and 0.3, already subpixel, I dont know how.

brunofavs, 2024-09-12:
These are the initial residuals from a calibration with all the sensors using the `dataset_corrected_with_odometry_and_depth_and_rgb` dataset.

Using a screenshot here to better highlight the colors. 

```bash
export CSF="lambda x: int(x) not in []" \
export DATASET=$ATOM_DATASETS/zau/inesc_day2_5_full/dataset_corrected_with_odometry_and_depth_and_rgb.json && rosrun atom_calibration calibrate \
-json $DATASET -uic -v  \
-csf "$CSF" -ssf "lambda x: x in ['rgb_body_left','rgb_body_right','rgbd_hand_color','rgbd_hand_depth','lidar_body']" \
 -ftol 1e-4 -xtol 1e-4 -gtol 1e-4 -phased

```

[image]

I don't think there is any collection that is so "outliar'y" to mess the final calibration.

> Can you run a calibration only with the body left and the hand camera?

Running this now.

miguelriemoliveira, 2024-09-13:
> One thing that is also bothering me is that the residuals are starting already way too small. In the `dataset_corrected_with_odometry_and_depth_and_rgb` in the calibration with 2 RGB for example, the residuals start at 0.9 and 0.3, already subpixel, I dont know how.

Yep. Also strange.

brunofavs, 2024-09-13:
So whats the plan now @miguelriemoliveira  @Kazadhum ? I see we're having overall decent results but there are still some things that we weren't expecting.

Should we meet to talk about this? I'm afraid to move on to the batches without polishing these edges out and end up wasting time like I did in the past. So I'm not sure how to proceed.

If so, I'm available today until 18h or tomorrow all day.

Kazadhum, 2024-09-14:
Hi @brunofavs! 

I'm available Monday morning to discuss things and see if we can find anything out. I agree that we should get these little problems out of the way before proceeding. Do you wanna meet at the lab or online?

brunofavs, 2024-09-14:
Lets meet Monday morning in the lab @Kazadhum 

Also I thought it was Thursday when I accidentally proposed a meeting Saturday :D

miguelriemoliveira, 2024-09-15:
Hi. Yes, we can talk tomorrow, but I am not sure at  what time. I will try to call in the morning, but I do not promise. Tomorrow is tricky...

@brunofavs , can you run again only the results that do not make sense just to make sure we are really having these problems?

Kazadhum, 2024-09-16:
Hi @miguelriemoliveira!

Me and @brunofavs have been discussing the results and after thinking a little bit I think these results are entirely plausible. I'll try to address each doubt in this issue as clearly as I can.

> About the results with the **dataset_corrected_with_odometry_and_depth.json**, overall they seem fine.
>
> However, I do not understand why when we use the rgb hand camera the results are ok (4 pixels [...]

I would argue that the results with the corrected initial guess for the RGB sensor are better. I don't think we can look only at the residuals from the `rgbd_hand_color` camera. After all, it is entirely possible that, with a bad initial guess, ATOM optimizes the sensor and pattern poses in such a way that the reprojection error *for that camera* is low, but not for all sensors. In fact, we can see that, when using the `dataset_corrected_with_odometry_and_depth_and_rgb` we get 14 pix and 9 pix errors for the `rgb_body_right/left` sensors while we get 11 pix and 7 pix for those same sensors after correcting the `rgbd_hand_color` sensor pose. So while the errors for that camera aren't significantly better, the errors for the entire system are. This seems entirely plausible to me.

> About the results with the dataset_corrected_with_odometry_and_depth_and_rgb.json, again overall they seem fine, but the rgb hand gives worse results than with the dataset_corrected_with_odometry_and_depth.json.
> Very strage ...

I don't think this is true right? Going from `dataset_corrected_with_odometry_and_depth.json` to `dataset_corrected_with_odometry_and_depth_and_rgb.json` we see the following:

- lidar_body: 0.0136 m -> 0.0136 m (stays the same)
- rgb_body_left: 14.2390 pix -> 11.6325 pix (improves)
- rgb_body_right: 9.1803 pix -> 7.1985 pix (improves)
- rgbd_hand_color: 5.3333 pix -> 5.0736 pix (improves slightly)
- rgbd_hand_depth: 0.0095 pix -> 0.0096 pix (practically the same)

So I'd say the results are better. 

> One thing that is also bothering me is that the residuals are starting already way too small. In the `dataset_corrected_with_odometry_and_depth_and_rgb` in the calibration with 2 RGB for example, the residuals start at 0.9 and 0.3, already subpixel, I dont know how.

I spoke with @brunofavs about this and it turns out this isn't true. It's just that the initial guess of the `rgbd_hand_color` generates a low reprojection error and only for that sensor. The other residuals being so high signals to me that the system is uncalibrated, which makes sense since we're only seeing the first iteration of the optimization.

---

All in all, I think the observations we're making aren't all that strange. Clearly there are some problems here, seeing as we're getting upwards of 7 pix of error for some sensors. We're investigating problematic collections now. But all in all I think these results make sense.

What do you think @miguelriemoliveira?

brunofavs, 2024-09-16:
Got a better result with less collections.

```
export CSF="lambda x: int(x)  < 30" \
export DATASET=$ATOM_DATASETS/zau/inesc_day2_5_full/dataset_corrected_with_odometry_and_depth_and_rgb.json && rosrun atom_calibration calibrate \
-json $DATASET -uic -v  \
-csf "$CSF" -ssf "lambda x: x in ['rgb_body_left','rgb_body_right','rgbd_hand_color','rgbd_hand_depth','lidar_body']" \
 -ftol 1e-4 -xtol 1e-4 -gtol 1e-4
```

```
+------------+----------------+--------------------+---------------------+----------------------+---------------------+
| Collection | lidar_body [m] | rgb_body_left [px] | rgb_body_right [px] | rgbd_hand_color [px] | rgbd_hand_depth [m] |
+------------+----------------+--------------------+---------------------+----------------------+---------------------+
|  Averages  |     0.0120     |       6.2644       |        9.7397       |        4.6925        |        0.0080       |
+------------+----------------+--------------------+---------------------+----------------------+---------------------+
```

Meaning there are probably some bad collections. Will investigate further.

brunofavs, 2024-09-16:
Hey,

Previously I was running calibrations with `ftol gtol xtol` set to `1e-4` and using all collections.

All the runs I'll be showing on this comment will be using the default `ftol gtol xtol` value.

I studied how the number of collections was affecting the calibration result. Below are results from the runs I did :

##### All collections

```
+------------+----------------+--------------------+---------------------+----------------------+---------------------+
| Collection | lidar_body [m] | rgb_body_left [px] | rgb_body_right [px] | rgbd_hand_color [px] | rgbd_hand_depth [m] |
+------------+----------------+--------------------+---------------------+----------------------+---------------------+
|  Averages  |     0.0171     |      100.4227      |       128.6394      |       257.6382       |        0.0151       |
+------------+----------------+--------------------+---------------------+----------------------+---------------------+

```
Although this results seem a lot worse, the previous ones were using `ftol gtol xtol = 1e-4`. I thought it would not affect the result this much.

##### Last 30
```
+------------+----------------+--------------------+---------------------+----------------------+---------------------+
| Collection | lidar_body [m] | rgb_body_left [px] | rgb_body_right [px] | rgbd_hand_color [px] | rgbd_hand_depth [m] |
+------------+----------------+--------------------+---------------------+----------------------+---------------------+
|  Averages  |     0.5990     | 272558499705217760.0000 |  1154222296338767616.0000 |  30928304605806864.0000 |        1.2778       |
+------------+----------------+-------------------------+---------------------------+-------------------------+---------------------+

```
##### First 30

```
+------------+----------------+--------------------+---------------------+----------------------+---------------------+
| Collection | lidar_body [m] | rgb_body_left [px] | rgb_body_right [px] | rgbd_hand_color [px] | rgbd_hand_depth [m] |
+------------+----------------+--------------------+---------------------+----------------------+---------------------+
|  Averages  |     0.0118     |       5.8897       |        9.8253       |        4.7468        |        0.0080       |
+------------+----------------+--------------------+---------------------+----------------------+---------------------+

```

The results got better from using only the first half of the total collections :

(In comparisson to the results from @Kazadhum 's previous comment)
> * lidar_body: 0.0136 m -> 0.0136 m (stays the same)
> * rgb_body_left: 14.2390 pix -> 11.6325 pix (improves)
> * rgb_body_right: 9.1803 pix -> 7.1985 pix (improves)
> * rgbd_hand_color: 5.3333 pix -> 5.0736 pix (improves slightly)
> * rgbd_hand_depth: 0.0095 pix -> 0.0096 pix (practically the same)

 - lidar_body:             0.0136 m     -> 0.0118  m   (improves slightly)
 - rgb_body_left:        11.6325 pix  -> 5.8897 pix (improves greatly)
 - rgb_body_right:      7.1985 pix    -> 9.8253 pix (worsens)
 - rgbd_hand_color:   5.0736 pix   -> 4.7468 pix (improves slightly)
 - rgbd_hand_depth:  0.0096 m    -> 0.0080 m   (improves slightly)

Total pixel error    23.9 pix -> 20.4pix

Then by reducing even further the number of collections it still gets a bit better : 

### Reducing the number of collections to 20

```
+------------+----------------+--------------------+---------------------+----------------------+---------------------+
| Collection | lidar_body [m] | rgb_body_left [px] | rgb_body_right [px] | rgbd_hand_color [px] | rgbd_hand_depth [m] |
+------------+----------------+--------------------+---------------------+----------------------+---------------------+
|  Averages  |     0.0137     |       5.6301       |        8.1524       |        5.2223        |        0.0086       |
+------------+----------------+--------------------+---------------------+----------------------+---------------------+
```

 lidar_body:              0.0118  m    -> 0.0137 m   (worsens slightly)
 rgb_body_left:         5.8897 pix  -> 5.6301 pix  (worsens slightly)
 rgb_body_right:       9.8253 pix  -> 8.1524 pix (improves)
 rgbd_hand_color:    4.7468 pix  -> 5.2223 (worsens slightly)
 rgbd_hand_depth:   0.0080 m   ->  0.0086(worsens slightly)

However the total pixel error : 20.4px -> 18.97px

This leads me believe that the odometry might not be still 100% fixed. Me and @Kazadhum played the dataset back and the pattern is indeed roughly static throughout the collections, leading to believe that the odometry would be accurate. However I'm seeing this tendency in my calibrations that so far I can't explain.

Also the runs with only the second half of the collections did not even converge, which also leads me to believe that the odometry is incorrect.

What are your thoughts on this @miguelriemoliveira ? From what I saw and understood the the method you developed seemed to follow the KISS ( keep it simple studid) ideology and seemed error-proof. I'm not sure what might be happening here.

closed 2024-09-20 by brunofavs