lardemua/atom #984: Initial pattern pose is calculated during data collection

Kazadhum, 2024-09-16:
Maybe it should be recalculated during calibration. This generated problems when calibrating the real Zau system.

Tagging @miguelriemoliveira and @brunofavs for visibility.

miguelriemoliveira, 2024-09-20:
After correcting the odometry, the pattern poses which are referenced w.r.t. the world frame, and are computed when the dataset is created, will be incorrect.

We need to create a script to recompute the pattern poses using the newly corrected odometry.

brunofavs, 2024-09-20:
And also change the calibrate script to compute these world-pattern transformations at runtime and to actually use them.

This new script you were refering to should only be for backwards compatibility.

miguelriemoliveira, 2024-09-20:
I don't think so but we can discuss further. Can you send me the dataset_corrected,json file? I accidentaly corrupted mine ...

brunofavs, 2024-09-20:
Give me a minute, I ll send you.

brunofavs, 2024-09-20:
[dataset_corrected.json](https://github.com/user-attachments/files/17073332/dataset_corrected.json)
[dataset_corrected_with_odometry_and_depth_and_rgb.json](https://github.com/user-attachments/files/17073334/dataset_corrected_with_odometry_and_depth_and_rgb.json)
[dataset_corrected_with_odometry_and_depth.json](https://github.com/user-attachments/files/17073338/dataset_corrected_with_odometry_and_depth.json)
[dataset_corrected_with_odometry.json](https://github.com/user-attachments/files/17073339/dataset_corrected_with_odometry.json)
[dataset_corrected_without_odometry.json](https://github.com/user-attachments/files/17073341/dataset_corrected_without_odometry.json)
[dataset.json](https://github.com/user-attachments/files/17073342/dataset.json)

Here are all the datasets I have, in case others get corrupted too.

miguelriemoliveira, 2024-09-20:
Can you come by the office?

brunofavs, 2024-09-20:
I'm not at the university right now. But we can go on zoom if you want.

miguelriemoliveira, 2024-09-20:
ok, lets zoom

brunofavs, 2024-09-20:
I think this can be closed now, but there is a slight nuance still.

One must now use the `estimate_pattern_poses` script to recompute the transformations after correcting the odometry of a dataset. So running this script is a necessary step for every single dataset with a mobile robot from now on.

This could automatically be done on the script that corrects odometry. It would be one less step to possibly forget and a more straightforward pipeline.

miguelriemoliveira, 2024-09-20:
> One must now use the estimate_pattern_poses script to recompute the transformations after correcting the odometry of a dataset. So running this script is a necessary step for every single dataset with a mobile robot from now on.

Well, only if you have a mobile robot and correct the odometry.

Thus, my suggestion is that you add a warning message to be printed alwasy at the end of the transform corrector (the odometry_inspector is depreated) saying be carefull to check if the pattern poses should not be recomputed.

miguelriemoliveira, 2024-09-20:
Can you do that?

brunofavs, 2024-09-20:
Yes I can.
Will do it this afternoon.

brunofavs, 2024-09-20:
63cad20581bd

Closing issue now.