lardemua/atom #1000: IMU Data Collection

Kazadhum, 2025-01-17:
Hi! For my research, I need to save data from IMUs in the ATOM dataset. 

The goal is to then be able to integrate this data to translate it into transformations. As such, a continuous stream of data should be saved. This implies a different system for data collection than used in ATOM thus far. IMU data cannot be saved in "collections" like camera images or LiDAR point clouds.

My current idea is to make use of the `additional_sesor_data` dataset field, changing it to accommodate the saving of continuous data. This change would be additive, so no functionality is lost and hopefully backwards compatibility is preserved. This should be tested when the PR is created.

Below is my TO-DO list for the implementation of this functionality, so it is documented. Development will take place in a separate branch.

---

## TO-DO:
- [ ] Create a new system under `atom_examples` called `rihibot` - which is a new version of `rihbot` with an IMU in the end-effector, for testing
- [ ] Add IMU modality to `configure_calibration_pkg` and related scripts
- [ ] Add checks to `configure_calibration_pkg` to verify sensor modality
- [ ] Implement `continuous` option for the collection of `additional_sensor_data`

This list will be updated as I progress and possibly find more obstacles.

Tagging @miguelriemoliveira @rarrais for visibility.

Kazadhum, 2025-01-20:
I just checked the [source code](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/959e7d02ecb8d0c8940623921220ce960542f6c7/gazebo_plugins/src/gazebo_ros_imu.cpp#L279C11-L279C35) for the `libgazebo_ros_imu` plugin and my previous assertion doesn't make sense... There is no place in the code where either `frameName` or `frame_id` are changed w.r.t. fixed TF's

miguelriemoliveira, 2025-01-20:
Hey @Kazadhum ,

not sure what is happening, but from what I gather the problem is not in atom right?

Can you point to the IMU plugin code?

Kazadhum, 2025-01-20:
Hello @miguelriemoliveira! Yeah, the problem isn't in ATOM, but in the gazebo plugin, I think. 

Here's the [gazebo plugin source code](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/959e7d02ecb8d0c8940623921220ce960542f6c7/gazebo_plugins/src/gazebo_ros_imu.cpp), and here's my use of the plugin in the URDF:

https://github.com/lardemua/atom/blob/ce1aa71f736bad63e14c98f50fd86b7a93470737/atom_examples/rihibot/rihibot_description/urdf/imu.urdf.xacro#L45-L72

Kazadhum, 2025-01-20:
If I delete those two lines, I get this error in gazebo:

`[FATAL] [1737370228.346497682, 3310.400000000]: imu plugin missing <bodyName>, cannot proceed`

which makes sense, given this:

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/959e7d02ecb8d0c8940623921220ce960542f6c7/gazebo_plugins/src/gazebo_ros_imu.cpp#L97-L100

Kazadhum, 2025-01-20:
I just checked and the same thing happens with `softbot`... Maybe I have to use the alternative `libgazebo_ros_imu_sensor` plugin? I never got it to work properly before, but I can try again.

Kazadhum, 2025-01-20:
Ok so I added the `libgazebo_ros_imu_sensor` plugin and I think it is working properly! The only thing to account for is that this plugin does measure gravity, so this must be considered in any algorithm that uses this data.

However, the IMU messages now have the proper `header_id`!

```
header: 
  seq: 9766
  stamp: 
    secs: 2722
    nsecs: 380000000
  frame_id: "imu_link"
orientation: 
  x: -0.0028736304093780474
  y: 0.03402877967051662
  z: -0.9956221813167769
  w: -0.08700722080625889
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.021119082716111827
  y: 0.003690111404028421
  z: 7.822583583077104e-05
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 0.11410727924185207
  y: -0.6591437516596792
  z: 9.777142752129746
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```

miguelriemoliveira, 2025-01-20:
> If I delete those two lines, I get this error in gazebo:
>
> `[FATAL] [1737370228.346497682, 3310.400000000]: imu plugin missing <bodyName>, cannot proceed`
>
> which makes sense, given this: [...]

right, but it proves those code lines are actually being used.

miguelriemoliveira, 2025-01-20:
> Ok so I added the `libgazebo_ros_imu_sensor` plugin and I think it is working properly! The only thing to account for is that this plugin does measure gravity, so this must be considered in any algorithm that uses this data.
> 
> However, the IMU messages now have the proper `header_id`!
> 
> ```
> header: 
>   seq: 9766
>   stamp: 
>     secs: 2722
>     nsecs: 380000000
>   frame_id: "imu_link"
> orientation: 
>   x: -0.0028736304093780474
>   y: 0.03402877967051662
>   z: -0.9956221813167769
>   w: -0.08700722080625889
> orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
> angular_velocity: 
>   x: 0.021119082716111827
>   y: 0.003690111404028421
>   z: 7.822583583077104e-05
> angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
> linear_acceleration: 
>   x: 0.11410727924185207
>   y: -0.6591437516596792
>   z: 9.777142752129746
> linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
> ```
> 
> So I think this roadblock is gone

Great news!

miguelriemoliveira, 2025-01-20:
> So I think this roadblock is gone

on to the next 10 million ...

Kazadhum, 2025-01-20:
Changed the `configure_calibration_pkg` script so no errors are found when it sees a sensor with the `imu` modality... A small note on this. I did not add RViz configurations for visualizing IMu data. The arrow that results from the use of that plugin can be very large depending on the linear acceleration. Keeping in mind that the simulated IMU measures gravity, the resulting arrow is very large and does not, in my mind, facilitate the visualization of data.

Moving onto the collection of IMU data...

Kazadhum, 2025-01-20:
Hi @miguelriemoliveira! Yes, that was what I meant!

When iterating through the sensor list, I added a `continue` when the modality is `imu`.

https://github.com/lardemua/atom/blob/448bc7023c2ec477099432425b6a3f8852375e22/atom_calibration/scripts/configure_calibration_pkg#L860-L908

But maybe this is not the proper way to do this. What do you think?

miguelriemoliveira, 2025-01-20:
> But maybe this is not the proper way to do this. What do you think?

Yep. Makes sense.

Kazadhum, 2025-01-22:
Hi @miguelriemoliveira and @rarrais! Continuous data collection was implemented! Check this the message for commit 7122ad1 for details.

I believe backwards compatibility was preserved with every change I made. I will test this further, however, before making the pull request.

I also added a warning (in commit c1d505c) for any user who includes an IMU to know that the scripts created during configuration might not be working properly if an IMU is included in the configuration.

The only two things missing is to add the `continuous` option to the `additional_sensor_data` field and to write the documentation for `rihibot`.

miguelriemoliveira, 2025-01-22:
Hey @Kazadhum ,

great work. 

Congratullations!

Kazadhum, 2025-02-05:
Closing this issue after merging PR #1010 ...

closed 2025-02-05 by Kazadhum